easyw/kicad-3d-models-in-freecad #1: Cannot generate 'gw' models

paulsykes, 2016-03-07:
I don't have any problems generating any of the other models; however I get the following error when trying to generate 'gw' models.

debug: Part.makeLoft catching 'Standard_Failure' msg: 'Need at least two vertices, edges or wires to create loft face'

Output from terminal:-

freecad make_gw_export_fc.py all

`FC Version 
['0', '15', '4671 (Git)', 'git://git.code.sf.net/p/free-cad/code', '2015/04/05 17:25:02', 'releases/FreeCAD-0-15', '244b3aef360841646cbfe80a1b225c8b39c8380c']FC Version 015
/usr/lib/freecad/Mod/CadQuery exists
FC Version 
['0', '15', '4671 (Git)', 'git://git.code.sf.net/p/free-cad/code', '2015/04/05 17:25:02', 'releases/FreeCAD-0-15', '244b3aef360841646cbfe80a1b225c8b39c8380c']FC Version 015
/usr/lib/freecad/Mod/CadQuery exists
Cannot find icon: :/icons/media-playback-start.svg
CadQuery 0.3.0
CadQuery is a parametric scripting language for creating and traversing CAD models
Author: David Cowden
License: LGPL
Website: https://github.com/dcowden/cadquery
Opened /usr/lib/freecad/Mod/CadQuery/Examples/Ex000_Introduction.py
Executed /usr/lib/freecad/Mod/CadQuery/Examples/Ex000_Introduction.py

Ex000 Closed
Dock adjusted

Running...

ACCE=11.8;E1=10.0;S=0.2;L=0.6
debug: Part.makeLoft catching 'Standard_Failure' msg: 'Need at least two vertices, edges or wires to create loft face'
Exception while processing file: make_gw_export_fc.py [Need at least two vertices, edges or wires to create loft face]`

Python console output:-
`Python 2.7.6 (default, Jun 22 2015, 18:01:27) 
[GCC 4.8.2] on linux2
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> import WebGui
>>> from StartPage import StartPage
>>> WebGui.openBrowserHTML(StartPage.handle(),'file://' + App.getResourceDir() + 'Mod/Start/StartPage/','Start page')
>>> import FreeCADGui as Gui
>>> App.setActiveDocument("Ex000_Introduction")
>>> App.ActiveDocument=App.getDocument("Ex000_Introduction")
>>> Gui.ActiveDocument=Gui.getDocument("Ex000_Introduction")
>>> App.setActiveDocument("")
>>> App.ActiveDocument=None
>>> Gui.ActiveDocument=None
>>> App.setActiveDocument("qfp52_10x10_p065")
>>> App.ActiveDocument=App.getDocument("qfp52_10x10_p065")
>>> Gui.ActiveDocument=Gui.getDocument("qfp52_10x10_p065")
>>> `

easyw, 2016-03-08:
at the moment you need to use CadQuery 0.2.0
I haven't updated my script to the new cadquery
Maurice

paulsykes, 2016-03-09:
Hi Maurice,
Thank you, everything working fine with CadQuery 0.2.0
Paul.

closed 2016-03-09 by paulsykes