easyw/kicad-3d-models-in-freecad #27: pinheaders not working

AnduriI, 2017-03-02:
I'm trying to create pin headers with your script (```c:/freecad/bin/freecad make_pinheaders_export_fc.py all```). But FreeCAD just gives an error message:
```
./generated_pinheaders/\PinHeader_Straight_1x02_H06_p1.27mm.step
./generated_pinheaders/Exception while processing file: make_pinheaders_export_fc.py [local variable 'LIST_license' referenced before assignment]
File format not supported: all 
```
and only created that one .step file. What am I doing wrong?

And a second question, is it possible to create angled pinheaders?

easyw, 2017-03-03:
@AnduriI 
fixing on the go

easyw, 2017-03-03:
ok, it should be fixed now
ths for reporting :)

closed 2017-03-03 by easyw