easyw/kicad-3d-models-in-freecad #28: The generator directory is a mess. How should we reorganize it?

poeschlr, 2017-03-03:
@easyw, @SchrodingersGat, @Shackmeister, @DASFrank, @metacollin:
How should the generators be organized.
For me 3 options come to mind: (sorted from most effort to least effort.)

Option 1:
Go for depth
[image]

Option 2:
My current effort. A bit of depth but also a lot of directories in the root folder.
[image]

Option 3:
Focus on a uniform repo
Effort by Oliver (as i understand it. I might have interpreted it wrong.) This is the solution that creates the least amount of work.
[image]

I vote for option 1 because i think it is the most future proof. (It will work even if we have 10 times the current generator scripts.)
Option 2 generates nearly the same amount of work as option 1 does. (Still has a cluttered root directory)
Option 3 could be a quick and dirty fix to get a uniform repository.

Any other suggestions?

easyw, 2017-03-03:
@poeschlr, @SchrodingersGat, @Shackmeister, @DASFrank, @metacollin 
I was thinking of something like that:
[image]
The structure will be similar to https://github.com/KiCad and when a family script generator is added, only one dir will be added to the repo. Output dir should be maintained empty on-line, but its structure is useful to test and generate models locally.
The reorganizing would be very easy but the result would be much cleaner than now...
What do you think?

poeschlr, 2017-03-04:
@easyw is my interpretation correct that what you suggest is very similar to option 2 but with the output directory moved to a separate directories within the root directory? 
By the way i like the mention of the _tools folder. I would do this this regardless of how the rest of the repo is organized.

SchrodingersGat, 2017-03-04:
Whichever way we go, we need to reduce the amount of duplicated files that are floating around. Currently it is a nightmare for maintenance.

e.g. a single license source script that gets copied into each file.

There's also much duplication of code which my PR https://github.com/easyw/kicad-3d-models-in-freecad/pull/24 tries to address.

The other thing to address is naming convention. Currently the different scripts have random naming and there is no need to have multiple .sh / .bat files that just serve to pass different arguments through...

easyw, 2017-03-04:
> e.g. a single license source script that gets copied into each file.

the new add_license.py is called with default license or with a custom one which is a variable in the main script

> The other thing to address is naming convention

about naming convention we could have for main files:
- main_generator.py 
- cq_parameters.py
- **create_model.bat**(sh) '**model_name**' or '**all**' to create all family models
other side files may be named with some grade of freedom
I like to have a single launcher inside the specific family folder to make the user approach to generation easier...
in **_tool** directory I would like to have:
- cq_cad_tools.py
- exportPartToVRML.py
- shaders.py
- also the 'cq_helpers.py' made by @poeschlr :D 
Version checking could be merged as function in cq_cad_tools.py
we could also add a DIR for '**screenshots**' of the generated models/family and a '**doc**' DIR for extra documentation
a **todo.md** and **readme.md** file will be at root to explain main usage and requirements
(FC 0.15 is best suggested for STEP generation and CQ 0.52 actual release which is tested ATM fine for all models)
Is this a good compromise for all of us?
Thx
Maurice

poeschlr, 2017-03-04:
Looks good

> create_model.bat(sh) 'model_name' or 'all'

This means i need to change my scripts.
(Currently i'm expecting filter=model_name or filter=*)
But this is easily done.

regarding the cq_helpers.py script: I need to play around a bit with that. currently i have a copy of it in all cq_models directories. I need to find a way to have a different python path for when the model is generated as standalone in the cadquery workbench and when it is generated from the export scripts.
(I'm sure there is a way. I was lacy when i made my last cleanup and i wanted to generate my models.)
I will look into this stuff today.

easyw, 2017-03-12:
@poeschlr  @SchrodingersGat 
I just started a refactory of the structure...
ATM I've done it for resistors_SMD... hte process to move the scripts to the new structure is very easy ...
I'm going to move all the scripts gradually...
Maurice

easyw, 2017-03-13:
capacitors_SMD done 😸

easyw, 2017-03-15:
capacitors_radial_SMD done  :grinning:

easyw, 2017-03-16:
DIP_packages done 😸

SchrodingersGat, 2017-03-16:
🥇

easyw, 2017-03-17:
GullWings QFP_SOIC_SSOP_TSSOP_SOT refactored 😸

easyw, 2017-03-21:
Pin Headers, Box Headers, Capacitors Radial THT completed! Only @poeschlr model scripts need to be slightly refactored to point to **_tools** and to generate in **_3Dmodels** sub dir ...

easyw, 2017-03-30:
@poeschlr  @SchrodingersGat 
you may condìsider useful this folder:
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/WIP
I added some WIP scripts...
1) option to generate molded QFN, and then BGA
[image]
2) option to load footprint beside the 3D generated model to easily check the result for all the family
(latest kicad StepUp Mod 4.1.2.8 needed)
[image]

easyw, 2017-04-17:
cleanup completed :)

closed 2017-04-17 by easyw