easyw/kicad-3d-models-in-freecad #45: Adding new tool?

hackscribble, 2017-04-22:
Hi @easyw 

For my recent models, I developed a Python class called `Ribbon`.  

https://github.com/hackscribble/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/Molex_55560/cq_models/ribbon.py

It let's me define a complex line as a series of straight line segments and arcs of given angle and radius, and then turns this centre line into a constant width closed wire (which can then be extruded to form a device pin or contact, for example).

Is this the sort of thing that might be added to the `_tools` folder, either as a standalone file or added into `cq_helpers.py`?  If so, I will raise a PR.

Thanks

Example of use:

[image]

easyw, 2017-04-22:
It would be perfect to be added to _tools dir...
Probably as a standalone file I think...
Thx for your additions!!!
Maurice :smile:

easyw, 2017-04-22:
would you mind also to have a look at this commit:
https://github.com/easyw/kicad-3d-models-in-freecad/commit/e69ec53cfd485dcfb90c4ea37b7212a007178623
I removed the old way to check if CQ is available, to be complaint with the add-on installer of FC that will place a folder named cadquery_module instead of CadQuery...
thx

hackscribble, 2017-04-22:
Thanks @easyw I will submit a PR to add my code.

About your commit - I have resynced from your master so that change is in my fork.  Did you want me to do anything else about it?

easyw, 2017-04-22:
your are fine Thx 😄

hackscribble, 2017-04-22:
Got it now - thanks :smile:  I will update that in my `main_generator.py` files.

Moving `ribbon.py` is not as easy as I thought.  Works OK when I run `main_generator.py` because it sets the relative path to `_tools`.  But when I run the model `.py` file from within CQ (as when developing and testing), FreeCAD copies it to a temp folder so the path doesn't work.  I can fix it by including a full path in the model file, but I don't like that solution :disappointed: 

So maybe I will leave `ribbon.py` where it is.

easyw, 2017-04-22:
Ok fine like that... No prob :)

closed 2017-04-22 by hackscribble