easyw/kicad-3d-models-in-freecad #48: Folder for standalone script

hackscribble, 2017-04-23:
Hi @easyw I have written a CQ script to generate the 3D models for a TO-220-7Lead_TabPin4 SMD device (aka TO-220-7-4).  It is a "standalone" script to be run from the CQ workbench in FC - it does not use the `main_generator.py` framework.

In `kicad-3d-models-in-freecad/cadquery/FCAD_script_generator/`, should I create a folder just for this script or create a generic folder `TO_SOT_Packages_SMD` to match the `.pretty` folder in the KiCad library?

Thanks

easyw, 2017-04-23:
Hi @hackscribble
A folder named TO_SOT_Packages_SMD already exists... But for custom models like yours I would suggest to create a special folder named TO_SOT_Packages_SMD_custom were collect all special scripts for parts non common ...
Then you can put the model in the same _3Dmodels subfolder TO_SOT_Packages_SMD of the common models 
Thx a lot for your nice work
Maurice

hackscribble, 2017-04-23:
Thanks, I will submit a PR soon.

closed 2017-04-23 by hackscribble