easyw/kicad-3d-models-in-freecad #50: Extending SMD TO and SOT scripts

hackscribble, 2017-04-23:
Hi @easyw and @Shackmeister 

I've opened this issue to continue the discussion from the PR ... https://github.com/easyw/kicad-3d-models-in-freecad/pull/49

>I'm adding Shackmeister to this conversation...
He did the work to extend sot packages to many asymmetric models and he was suggesting me to improve the generation for other models - @easyw 

> I was considering adding and option to set each pin induvidualy to either a GW or a flat pins... also I was thinking about adding an option in exclude_pins to set a value to 2.5 to only have half of the pin. would be used for TO-252 etc. - @Shackmeister 

Can you confirm that the existing scripts are in this folder?  
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT

Thanks

easyw, 2017-04-23:
@hackscribble 
yes here they are :smile:

hackscribble, 2017-04-24:
### Building a list of possible devices to include.  
I will update this comment as I find new ones.

Package | Aliases | Image
---|---|---
TO-268 | D3PAK |  [image] 
TO-263-7 | D2PAK7 | [image] 
TO-263-7 centre lead trimmed | D2PAK7-T, SOT427 | [image]
TO-263-5 | D2PAK5 | [image] 
TO-263AA | D2PAK3 | [image] 
TO-263AB | D2PAK | [image] 
TO-252 | DPAK, TO-252AA | [image] 
TO-243AA | SOT-89 | [image] 
TO-252-3L | DPAK3-T | [image]
TO-252-5 | DPAK5-T | [image]
TO-252-7F | DPAK7F-T | [image]
TO-263-4 center lead trimmed | SOT-426 | [image]

hackscribble, 2017-04-24:
@easyw @Shackmeister If you know of other packages that should be added to the list, please suggest them.  I will do some more searching over the next couple of days and then start to collect drawings / dimensions.

hackscribble, 2017-04-24:
Looks like most of the packages in the earlier comment boil down to the 11 models in the tables below.  This is where I am going to focus.

There are some gaps in `TO_SOT_Packages_SMD.pretty` so my first job is to create the missing footprints and check that the existing ones are correct.

### DPAK

| Pins | Cropped pin | Aliases                   | KiCad footprint     |
|------|-------------|---------------------------|----------------------|
| 3    | no          | TO-252-3L                 | x                    |
| 3    | yes         | TO-252AA, SOT-428         | SOT-428              |
| 5    | no          | TO-252-5                  | x                    |
| 5    | yes         | TO-252-5-11               | TO-252-5-11          |

### D2PAK

| Pins | Cropped pin | Aliases                   | KiCad footprint     |
|------|-------------|---------------------------|----------------------|
| 3    | no          | TO-263AA                  | TO-263-3Lead         |
| 3    | yes         | TO-263AB, SOT-404         | SOT-404              |
| 5    | no          | TO-263-5                  | TO-263-5Lead         |
| 5    | yes         | TO-263-4, SOT-426         | x                    |
| 7    | no          | TO-263-7, TO-220-7-4      | TO-220-7Lead_TabPin8 |
| 7    | yes         | TO-263-7 cropped, SOT-427 | x                    |

### D3PAK

| Pins | Cropped pin | Aliases                   | KiCad footprint     |
|------|-------------|---------------------------|----------------------|
| 3    | yes         | TO-268                    | x                    |

easyw, 2017-04-24:
@hackscribble I'm adding @SchrodingersGat to this conversation...
looking at the models you are evaluating to create with your scripts I would consider your list exhaustive...
Regarding the model already present I would vote to substitute all with the ones coming from your scripts  ... that will give an uniformity to the library (same style, color as per kiCAD materials etc) and also a affordable and easy way to modify / update if needed...
@SchrodingersGat what do you think?
Thx
Maurice

SchrodingersGat, 2017-04-24:
Agreed, it's not too much effort to add the dimensions for existing models to a script, and that way we have consistency and also the ability to update all models at once

hackscribble, 2017-04-26:
PR submitted for DPAK, D2PAK and D3PAK footprints: https://github.com/KiCad/TO_SOT_Packages_SMD.pretty/pull/47

Once I get feedback on them and finalise them, I will work on the 3D models for them.

hackscribble, 2017-04-28:
Still work to be done on the pins, the tab shape for TO263 (D2PAK) and generally on TO268 (D3PAK) but the models are coming along nicely.

[image]

Shackmeister, 2017-04-28:
Wow, that is awesome man! I'm a huge fan :p thanks for your work

easyw, 2017-04-28:
Great! Second fan here! ;)

easyw, 2017-05-01:
I think now we can close it :smile:

closed 2017-05-01 by easyw

SchrodingersGat, 2017-05-01:
This is looking really good :)

Shackmeister, 2017-05-08:
hey @hackscribble can you please consider adding the following models:
SOT669
TO252-2

hackscribble, 2017-05-08:
Hi @Shackmeister 

TO-252-2 is among the models that were merged into the library: https://github.com/KiCad/kicad-library/pull/1204 :smile: 

Happy to add SOT-669 but I won't have time until next week.

Shackmeister, 2017-05-08:
changed the TO-252-2-Rectifier to use the same model :)

Shackmeister, 2017-05-08:
perhaps yours is better :P not sure though :package: 
https://forum.kicad.info/t/3d-new-library-for-mechanical-cad-exporting-and-enclosure-design/1763/316
can you please check if other of the missing ones could be made with your script? :)

hackscribble, 2017-05-08:
From all the current footprints in `TO_SOT_Packages_SMD.pretty`, I've extracted the unique 3D model names.

These are the models which do **not** exist in `TO_SOT_Packages_SMD.3dshapes`, so I will see if can model any of these.

[image]

There are also several 3D models that are not referenced any more by a footprint ...

[image]

Shackmeister, 2017-05-08:
I already added quite a few of those you listed :) I'll add them in the repo when @maui has had a look at them :)

Shackmeister, 2017-05-08:
Or you can see the list of freshly added ones here 
https://github.com/easyw/kicad-3d-models-in-freecad/pull/59/files

hackscribble, 2017-05-08:
If I'm reading your PR right, the only missing ones will be ...

[image]

I can do the first and last with my current script. The SOT-133n would need a different script. 

About the PQFN:
- maybe it belongs in `Housings_QFN_DFN`?
- so far, I can't find a device that has a pinout / recommended land pattern that matches the footprint :smile:

Shackmeister, 2017-05-08:
I made the 1333 and 1334 with the QFN script (the ground pins are not connected all the way through but that shouldn't be too much of an issue)

Shackmeister, 2017-05-08:
If you can fix those I'll brew up a script to make the To50 /powermacro models :)

easyw, 2017-05-08:
for the 1333 I found a workaround 😜  through the epad 😉

Shackmeister, 2017-05-08:
Screenshot? :)

easyw, 2017-05-08:
[image]
💤 😴 😉

Shackmeister, 2017-05-08:
Nice :p can you please check the models and perhaps the script when you have time? :)

easyw, 2017-05-09:
I will
Thx

hackscribble, 2017-05-15:
There was a commit two days ago to the 3D models for `TO_SOT_Packages_SMD.pretty` ...

https://github.com/KiCad/kicad-library/commit/a3e3dcaf2cee73bb0ad9e06d32805237606908e4

Running my check program against the footprints and models now shows these gaps ...

```
Checking TO_SOT_Packages_SMD (contains 98 modules)
- No 3D model for reference Infineon_PG-HSOF-8-1.wrl in module Infineon_PG-HSOF-8-1.kicad_mod
- No 3D model for reference Infineon_PG-HSOF-8-1.wrl in module Infineon_PG-HSOF-8-1_ThermalVias.kicad_mod
- No 3D model for reference PQFN_8x8.wrl in module PQFN_8x8.kicad_mod
- No 3D model for reference SOT-669_LFPAK.wrl in module SOT-669_LFPAK.kicad_mod
- No 3D model for reference SOT-669_LFPAK_ThermalVias-1.wrl in module SOT-669_LFPAK_ThermalVias-1.kicad_mod
- No 3D model for reference SOT-669_LFPAK_ThermalVias-2.wrl in module SOT-669_LFPAK_ThermalVias-2.kicad_mod
- No 3D model for reference SOT89-3_Housing.wrl in module SOT89-3_Housing.kicad_mod
- No 3D model for reference SOT89-3_Housing_Handsoldering.wrl in module SOT89-3_Housing_Handsoldering.kicad_mod
- No 3D model for reference SOT89-5_Housing.wrl in module SOT89-5_Housing.kicad_mod
- No 3D model for reference SOT89-5_Housing_Handsoldering.wrl in module SOT89-5_Housing_Handsoldering.kicad_mod
- No 3D model for reference TO-252-2_Rectifier.wrl in module TO-252-2_Rectifier.kicad_mod
- No 3D model for reference TO-277A.wrl in module TO-277A.kicad_mod
- 12 module warnings
```

The `TO-220-7Lead_TabPin8` now has a model, so I will drop that from my to do list.  But the `Infineon_PG-HSOF-8-1` and `SOT-669_LFPAK` still need models, so I will add those to my script.

I will also submit a PR for the footprint repo so that the variants (e.g. normal vs different vias) all point to the same 3D model.

Shackmeister, 2017-05-15:
sounds good :)

Shackmeister, 2017-05-15:
you should probably check against packages3D instead of kicad-library (packages3D should hopefully not have any og the old wings WRL files)

Shackmeister, 2017-05-15:
unless your script can handle the TO-50 I can make one :)

easyw, 2017-05-15:
@hackscribble is there possible to see your checking script? this sounds nice to have! 😄

hackscribble, 2017-05-15:
@Shackmeister my current script won't do TO-50, it would need me to write a new script.

@easyw sure thing ... https://github.com/hackscribble/kicad-library-utils/blob/add_check_3d_coverage/pcb/check_3d_coverage.py  I have submitted a PR in case the librarians want to add it to the kicad-library-utils.  The current version needs editing to change to `packages3D` repo from `kicad-library` repo but I will make that a command line argument in the next version, once I get any feedback generally on it from the librarians.

hackscribble, 2017-05-15:
SOT-669 done.  I need to submit a PR for the footprint to remove the old position and scaling values.

[image]

Shackmeister, 2017-05-15:
@hackscribble from what I can see in your scriptchecking the .Pretty foldet is outdated, you should probably grab the newest and rerun it

hackscribble, 2017-05-15:
I thought I pulled the latest yesterday, but I'll double check.

Shackmeister, 2017-05-15:
kicad-library is filled with all the old models while packages3D only have the new models which also have the step part. So if possible all the models in kicad-library should be remade in freecad :)

hackscribble, 2017-05-21:
Pr submitted for tidying up the SOT-669 and HSOF-8-1 footprints: https://github.com/KiCad/TO_SOT_Packages_SMD.pretty/pull/54

That leaves the SOT-89-3 model to finish.