easyw/kicad-3d-models-in-freecad #112: Scripts you want made

grob6000, 2017-10-11:
Hoping I could contribute some additional scripts. Promise to be tidier than the IDC headers this time! Let me know what you need next and I'll have a crack. Otherwise will be doing one to generate some TO92s (which were missing from my last project...).

easyw, 2017-10-12:
Hi,
thx  a lot for your contribution!
TO92s is fine though! 
We are missing also TH resistors if you want to go with...
Thx 😸 
Maurice

grob6000, 2017-10-12:
Resistors it is. Starting now. Using pin headers as example as this seems
to have a good config setup system!

Shackmeister, 2017-10-14:
Things I would like to see:
SMD Inductors!! WE_MAPI especially :)
THT TO_SOT packages
THT reisistors
THT LEDs
Pin Sockets
JST_GH connectors
Molex Clikmate connectors
Molex Duraclik connectors
AC/DC and DC/DC converters
Relays SMD andT
LDR resistors
Terminal blocks
Dip switches, SMD and THT
SMD crystals and Oscilattors

easyw, 2017-10-14:
> and it cannot generate a STEP file without the fusion

for THT resistors there is also a small problem with the Sketch not fully constrained that will generate a wrong STEP file

Shackmeister, 2017-10-14:
Wasn't that with Kammutierspules resistors?

easyw, 2017-10-14:
not only... I made only few test with TH resistors, but if you open the STEP exported i.e. in Varicad, you will see the errors...
i.e. the R_Axial_DIN0204_L3.6mm_D1.6mm_P5.08mm_Horizontal.FCStd
[image]
[image]

Shackmeister, 2017-10-14:
Should be ported to cadquery anyway :p

Shackmeister, 2017-10-20:
@grob6000 are you working on the THT resistors?

grob6000, 2017-10-20:
I have been, have a script that makes a reasonable-looking version of most
of them now. Slightly simplified relative to the original WRLs but I'm not
sure whether that might be a good thing. Robust discussion welcome! :)

It doesn't do the arrays yet, and I haven't verified some of the parameters
properly (likely should reference a datasheet for each size; used Vishay
for some things already).

See
https://github.com/grob6000/kicad-3d-models-in-freecad/tree/resistors_tht

easyw, 2017-10-20:
very nice though!
It would be possible to add a fillet like in this picture?
[image]

and if I remember correctly Mario suggested "light brown body" as color for resistors
[wrl materials](https://cld.pt/dl/download/64e39e99-c5b6-451b-accd-9e25331ceafe/KiCad_3D-Viewer_component-materials-reference-list_MarioLuzeiro.pdf?download=true)

Shackmeister, 2017-10-21:
Awesome :) ill be digging into the SMD inductors soon :)

grob6000, 2017-10-21:
Brown it is, and I’ll put the fillet in. :)

easyw, 2017-10-21:
>  I haven't verified some of the parameters properly 

if you want to easily check the model over its footprint you can add some code to load the fp in FC while generating the model.
Please have a look at i.e BGA generator ...

terjeio, 2017-10-25:
@Shackmeister you mentioned DIP switches above. As I have just made a script for DIP sockets and since I am a newbie to pyton I wanted to learn more about the language. I am in the process of recoding my script to utilize object oriented programming and I have started to add other DIP parts as  "part" objects to verify my approach, the first beeing slide type DIP switches. As these has text/numbers on the body (ON and switch number) I wonder if anyone has successfully added text to the body via CadQuery and if so how and which font to use. Or should the text be left out to reduce file size?

[image]

Shackmeister, 2017-10-25:
@terjeio That is really nice looking! is it made for the Kicad footprints?
I havent made anything with text on the model, also adds too much size for my taste.
You could consider making a small triangle for pin 1 and leave it like that :)
Make sure to upload the models to the Kicad Library when you are finished :D 
can you also make the SMD version? :)

terjeio, 2017-10-25:
@Shackmeister It is for the Kicad footprints and I have a SMD version in the pipeline. Triangle it will be and I am going to upload when I am happy with the script.

terjeio, 2017-10-26:
@Shackmeister During QC of my footprints I see that the THT DIP-sockets has different models assigned for "normal" version and LongPads version. Maybe I am a bit thick  but aren't the models supposed to be the same? If not what are the (physical) differences?

Shackmeister, 2017-10-26:
I strongly believe both should refer to the same model ("normal"). I noticed the kicad-library is still containing the old wings models. Can you please make sure both are up to date and has all the "possible" models?

terjeio, 2017-10-27:
I see that the footprints are already updated - I need to update more often...
I now have 126 models ready for review, next step is to learn how to make a PR.

[image]

Shackmeister, 2017-10-27:
You should always update footprints before adding models :) would you like some tips on the kicad PR? Or is it the git stuff in general which is trouble some? :)

BTW those models look really sweet! Keep up the good stuff!

Misca1234, 2017-10-28:
Oh, it seems that we have made the same work in parallel without knowing each other work, i just pushed some 3D models for DIP switched SMD
https://github.com/KiCad/kicad-packages3D/pull/151

Fell free to overwrite mine

Misca1234, 2017-10-28:
@Shackmeister
About the THT LEDs, in kicad-library there are extensive amount of .wrl files for THT Leds, but they have not been copied to the package3D repository

A couple are made in wings but they are few

Shackmeister, 2017-10-28:
@misca1234 these are made by Jan with a Freecad script but not CQ.. For this reason they don't make step models, correctly scaled WRL or step models at all... My take for the caps made by him was simply to rewrite them to CQ. After this I modified his footprint generator to output the correct parameters

Misca1234, 2017-10-29:
Ok

Misca1234, 2017-10-29:
FYI, so we don't get efforts made in parallel, I am currently fixing 3D models for Coverters_ACDC_DCDC

terjeio, 2017-10-29:
And I am working on socket strips - coding is nearly finished, need to verify parameters.

[image]

easyw, 2017-10-29:
wow ... what a great MCAD team we are getting!
Thanks all for your great job!
Maurice

Shackmeister, 2017-10-29:
Great job guys! @terjeio I did a pretty big overhaul of the pinheader footprints, can you have a look at the socket strips footprints or should I? Also we should prepare ourself for people going apesh*t if we add 2000 models more for the socket strips

terjeio, 2017-10-29:
@Shackmeister I believe I have the code ready for the socket strips now, but need to fine tune pin-parameters. A previous test run generated over 1GB...

What to do about the footprints for these? They have model-parametes (offset and rotation) incompatible with StepUp and there are many. Do you know if they have been generated by a script that could be rerun?

Another example, too much details?...

[image]

...maybe not, your pin headers folder is 1.4GB.

Shackmeister, 2017-10-29:
any kind of rotation, scaling or offset should be done in the 3D mode, not the footprint! they should always be  111 for scaling, 000 for rotation and 000 for offset.
I love the high detail level, I think we should deal with the size issue in another way than cutting down the quality.

footprint script can be found here, please let me know if you need some help :)
https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/pin-headers_socket-strips
the footprints will probably need some adjustment. but its much easier to use to change the wrong scalings

terjeio, 2017-10-30:
@Shackmeister I think you misunderstand me, the footprints do not have standard values, for example:

   (model ${KISYS3DMOD}/Socket_Strips.3dshapes/Socket_Strip_Angled_2x31_Pitch2.00mm.wrl
    (at (xyz 0.03937 -1.181102 0))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 90))
  )

  (model ${KISYS3DMOD}/Socket_Strips.3dshapes/Socket_Strip_Angled_1x20_Pitch2.00mm.wrl
    (at (xyz 0 -0.748031 0))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 90))
  )

So, should I adjust the generated models accordingly or should the footprints be changed? I believe the footprints should be changed to comply with the standard convention.

Shackmeister, 2017-10-30:
I didnt misunderstand you :)
All footprints should be changed like so:
(model ${KISYS3DMOD}/Socket_Strips.3dshapes/Socket_Strip_Angled_1x20_Pitch2.00mm.wrl
(at (xyz 0 0 0))
(scale (xyz 1 1 1))
(rotate (xyz 0 0 0))
)

From here the 3D model should be fitted to the footprint

terjeio, 2017-10-30:
Ok, the easiest way would be to rerun the script then. Should I do that and make a PR for the footprints also? Or is this something that the librarians takes care of?

For the models: when a PR is merged for scripts who makes a PR for them?

Maybe the workflow is described somewhere, but for now I am a bit overwhelmed by all the details surrounding this.

Shackmeister, 2017-10-30:
If you can fix the footprints it would be awesome, if not I can probably fix it :) 
Change the script part which makes the offset on the model to 000, the scaling to 111 etc.
Make a PR on the library and note this shouldn't be merged until the new models are ready (since they break compatability with the current models). From here you should have the stuff you need to finish the 3D models. 
Once you finished these you should generate the models and make a PR in kicad-library and packages3D (remember screenshots). And make a PR in this to make sure your script is kept here for future revisions. 
From here a Librarian (possibly me) will check your PR's and review it (and possibly specify some needed changes. 

Once the librarian is satisfied the footprint and 3D models will be merged :)

easyw, 2017-10-30:
@Shackmeister 

> Also we should prepare ourself for people going apesh*t if we add 2000 models more for the socket strips

you will find users complain also for missing footprints ... it is impossible to cover all users needs ...
I would consider better to have a ready to go library and not asking users to build they own models from scripts... The big issue IMO is to create a way to download only needed 3D parts...

An interesting option would be to have a function to be called inside Kicad or pcbnew to create a full list of needed 3D parts for the user kicad project, then check automatically if those models are already in the users local 3D library and give an option to retrieve with a 'wget' process the missing models from kicad on-line library; finally the downloaded missing models would be placed automatically in the right local folder by the function itself.
For example if you need:
'Buttons_Switches_SMD.3dshapes/SW_SP3T_PCM13.step'
your wget could point to:
https://github.com/KiCad/kicad-packages3D/raw/master/Buttons_Switches_SMD.3dshapes/SW_SP3T_PCM13.step1
and download it directly, placing it to local configured
'KISYS3DMOD'+Buttons_Switches_SMD.3dshapes/
folder.
This could be done as a python Action Script for Kicad

Shackmeister, 2017-10-30:
@easyw I couldn't agree more! My point was merely we should prepare :)

terjeio, 2017-10-31:
@Shackmeister 
> If you can fix the footprints it would be awesome, if not I can probably fix it :)

I have been looking into this and it is a mess, the script you referred to has been changed, most likely after it was committed, and is broken. From what I can glean from the master footprints it is higly likely that a variant of this script has been used to generate the footprints since many matches exactly.  

I have managed to correct the scaling and rotation but there are other errors as well, for example:

[image]

This is *Socket_Strip_Straight_2x10_Pitch2.54mm_SMD.kicad_mod*, as you can see the pads are overlapping.

I do not think I want to try to fix the script as there are other changes as well, maybe the correct way forward is to raise an issue for the source?

BTW the models are ready for review, I had to take most of the measurements from the footprints though as there are a zillion variants available.

antoniovazquezblanco, 2017-11-04:
Wow guys. I wanted to congratulate you on the work done. 3D library is looking nicer every day.

I miss FPC connector 3D models. Particularly Hirose FH12 series. I made a PR for a single connector footprint but I plan to submit an scripted generated version of the family when I find some time for investigating the footprint generation scripts.

Thank you!

terjeio, 2017-11-04:
@antoniovazquezblanco 

I have started work on Molex 52271 and in the process I am adding a class based approach to scripting the models since I like reusing code. One class is for polylines, due to I find it cumbersome to work in absolute coordinates, it allows code like this:

```phyton
        body = Polyline(cq.Workplane("XY"))\
                    .addPoint(0,1)\
                    .addThreePointArc((0.5, -0.2), (1, 0))\
                    .addPoint(0,-1).make().extrude(self.body_height)
```
and this (for the ribbon inside the body in the example below):
```phyton
        r = Polyline(cq.Workplane("XZ").workplane(offset=-self.body_width / 2.0), origin=(-self.cable_slot / 2 + 1.0 , self.body_height / 2))

        r.addPoint(0.0, 0.5)
        r.addPoint(0.8, 0.0)

        for i in range(1, self.num_pins):
            r.addPoint(0.0, -self.pin_thickness)
            r.addPoint(self.pin_width, 0.0)
            r.addPoint(0.0, self.pin_thickness)
            r.addPoint(1.0 - self.pin_width, 0.0)

        r.addPoint(0.8, 0.0)
        r.addPoint(0.0, -0.5)

        int = r.addMirror("Y").make().extrude(self.body_width)   
```
Example from work in progress for the Molex connector
[image]

I have wrapped lot of  other common stuff in classes too so when making new models one need only to concentrate on the parameters an the model itself when making a new. 
 
The code is work in progress and I will use for my next PR, I can upload a preliminary version if of interest.

antoniovazquezblanco, 2017-11-05:
@terjeio Nice models. I like the object oriented approach but first I will focus on learning how to script the footprints and in the future will take a look at 3D models ;)

poeschlr, 2017-12-11:
@terjeio your suggested extensions for cadquery might also be welcome by the cadquery project it self.

antoniovazquezblanco, 2018-09-25:
Hi everyone, although I wanted to learn more about the 3D scripts you make guys I never found the time to do it. I am currently trying to help a little bit wih KiCad by organizing and improving the situation of the pending merges in the oficial repos.

I am here to ask for some models that could help me save time at work that I can later invest on helping with more librarian tasks. Can anybody try and script `Connector_Harwin.pretty/Harwin_M20*` that I am currently using?

I will try to review the models myself with the highest priority and will invest some more time solving more issues at the bug tracker ;).

Thanks!

Misca1234, 2018-09-25:
> I wanted to learn more about the 3D scripts you make guys I never found the time to do it

For example, one place to start with is 

https://github.com/easyw/kicad-3d-models-in-freecad/cadquery/FCAD_script_generator/molex/cq_models/conn_molex_picoflex_90814.py

in directory 
https://github.com/easyw/kicad-3d-models-in-freecad/cadquery/FCAD_script_generator/mole

you then feed Freecad with "main" python scripts 

...../FreeCAD.exe main_generator.py series=90814

> Can anybody try and script Connector_Harwin.pretty/Harwin_M20* that I am currently using?

Allow me to try to fix them, give me a couple days