easyw/kicad-3d-models-in-freecad #120: align the License of the 3D models to the kicad libraries updated License

easyw, 2017-11-26:
Hello contributors.

This issue is raised to align the License of the 3D models to the kicad libraries updated License.

I'm kindly asking all the contributors to agree with this new license from GPL with symbol exception to CC-BY-SA 4.0  with symbol exception, as it is now adopted in KiCad libraries.
Here the new License file included into the models generator:
[aligning License to CC-BY-SA 4.0](https://github.com/easyw/kicad-3d-models-in-freecad/commit/fc8134933cb24e69f239c035dd86dc7ed9a55ac9)
I'm adding the list of all contributors:
@hackscribble @hyOzd @DASFrank @poeschlr @stoth @Misca1234 @timtashpulatov @grob6000 @metacollin @terjeio @SchrodingersGat @Shackmeister @patricklaf @joe-skb7 

Thanks a lot for your contribution.
Maurice

SchrodingersGat, 2017-11-26:
Thanks @easyw for raising this issue. I consent to such a license change to any and all of my contributions

Shackmeister, 2017-11-26:
Awesome! I Agree to have all my models relicensed in any way :)

hyOzd, 2017-11-26:
Does this change includes the license of the generator scripts? Can you clarify?

easyw, 2017-11-26:
@hyOzd 
this is to change the License of the 3D models, not of the generators...
I added a clarification on the first comment.

Shackmeister, 2017-11-26:
Will this be updated in stepup as well?

easyw, 2017-11-26:
@Shackmeister 
> Will this be updated in stepup as well?

StepUp tools don't add any particular License to the models exported manually...
Anyway the License in general is the one assigned to the Library from which the model is downloaded.

joe-skb7, 2017-11-26:
I agree, you can re-license my model.

hyOzd, 2017-11-26:
@easyw thanks for clarification. I agree with license change.

Shackmeister, 2017-11-26:
@easyw so what's the proper way to add the license from a manual made model?

easyw, 2017-11-26:
@Shackmeister 
WRL readers don't read any license at all; the license is injected as a comment inside
STEP readers have some fields reserved for:

- author
- organization
- authorization

they can be edited manually with a text editor... moreover not all the STEP readers are reading those and in many case they just overwrite them...
Then the License is important as stated at the source library ...

terjeio, 2017-11-26:
Great, but isn't the wording somewhat misleading?
"these libraries" is referring to a collection of files not?  The license text is to be injected into every single file and, in my opinion, it would be more precise to bind the license to the file itself - not some outside entity. Maybe "this library file" or "this file" is better than "these libraries"?

English is not my first language and I am not very well versed in legalese so sorry if I am wrong.

easyw, 2017-11-26:
> Great, but isn't the wording somewhat misleading?

The main purpose is to change the License from GPL with exception to CC with the same exception...
These files will be part of the KiCad libraries and the License will be passed to them as a collection of single files.
English is not my first language either, so I'm open to some suggestion, but one has to remember that the License is for the generated files and it is injected on the files themselves as an additional optional condition...
Moreover the allowed number of characters inside the STEP file header are limited, so I had to update the text adding a link to the KiCad original License and dropping a part of the full text license...
Maurice

terjeio, 2017-11-26:
This wording is what I was tinkering with moments before your initial mail arrived:

```
                "This file is licensed under the Creative Commons CC-BY-SA 4.0 License with the following exception:",
                "To the extent that the creation of electronic designs that use 'Licensed Material'",
                "can be considered to be 'Adapted Material', then the copyright holder waives",
                "article 3 of the license with respect to these designs and any generated files",
                "which use data provided as part of the 'Licensed Material'.",
                "",
                "https://creativecommons.org/licenses/by-sa/4.0/legalcode",
                "---",
                "",
                "Risk disclaimer:",
                "*USE 3D CAD DATA AT YOUR OWN RISK*",
                "*DO NOT RELY UPON ANY INFORMATION FOUND HERE WITHOUT INDEPENDENT VERIFICATION.*",
                ]
``` 

There is absolutely nothing that stops anybody from distributing and using the model files outside of the KiCad framework, and there is no mechanism to keep them bundled together.  So I still believe it is better to bind the license, which is exactly the same as used by KiCad, to the file itself to avoid ambiguity.
I would not be surprised if files find their way into other libraries over time...

Some references:
[Marking your work with a CC license](https://wiki.creativecommons.org/wiki/Marking_your_work_with_a_CC_license)
[Best practices for attribution ](https://wiki.creativecommons.org/wiki/Best_practices_for_attribution#This_is_a_good_attribution_for_material_you_modified_slightly)

Terje

grob6000, 2017-11-26:
Fine with any models I’ve made being licensed however is necessary.

easyw, 2017-11-26:
@terjeio 

> This wording is what I was tinkering

I just modified the first sentence to:

> **this file and libraries** are licensed under the [Creative Commons CC-BY-SA 4.0 License](https://creativecommons.org/licenses/by-sa/4.0/legalcode), 
> with the following exception:

and I already said I added a link to the KiCad full License
> Full text of the License here: https://github.com/KiCad/kicad-packages3D/blob/master/LICENSE.md

I think this is should be enough, but I'm open to other suggestion from KiCad developers...

terjeio, 2017-11-26:
@easyw 
I am going to add the license information like in this example:
```
/* SW_DIP_x5_W7.62mm_Slide.step 3D STEP model for use in ECAD systems
 * Copyright (C) 2017, Terje Io
 * ---
 * This work is licensed under the Creative Commons CC-BY-SA 4.0 License with the following exception:
 * To the extent that the creation of electronic designs that use 'Licensed Material'
 * can be considered to be 'Adapted Material', then the copyright holder waives
 * article 3 of the license with respect to these designs and any generated files
 * which use data provided as part of the 'Licensed Material'.
 * 
 * https://creativecommons.org/licenses/by-sa/4.0/legalcode
 * ---
 * Please refer to http://kicad-pcb.org/libraries/license/ for further clarification of the exception.
 * ---
 * 
 * Risk disclaimer:
 * *USE 3D CAD DATA AT YOUR OWN RISK*
 * *DO NOT RELY UPON ANY INFORMATION FOUND HERE WITHOUT INDEPENDENT VERIFICATION.*
```
The text "What does this mean?" and below in the "full" license is not part of the license itself as I read it so I opted to add a link to keep it short.

As a side note: the two first sentences in last paragraph for "What does this mean?" seems for me to contradict each other - but that could be due to my limited understanding of english: "the exception above does not apply" vs "shared under the same license agreement". Or is the intention that I give KiCad some exclusive rights to my work? If so perhaps the text in the first line should be changed: "...3D STEP model for *use in ECAD systems*..."

Terje

easyw, 2017-11-26:
Hi @terjeio 
I've added few small changes as per your suggestion...
You can have a look at the full text here:
https://github.com/easyw/kicad-3d-models-in-freecad/blob/9814e0b63f05c32665da5ffac43f3b0b4a09cbc9/cadquery/FCAD_script_generator/_tools/add_license.py
Maurice

patricklaf, 2017-11-27:
I agree this new license.

hackscribble, 2017-12-04:
I agree to this new licence.