easyw/kicad-3d-models-in-freecad #122: SW DIP models rotated

antoniovazquezblanco, 2017-11-29:
[image]

The models for the SW DIP seem to have been exported rotated around PIN number 1 of the component.

Sorry I cannot find the time to guess why I cant execute the scripts with my version of FreeCAD and CadQuery.

Can anyone have a look at this?

easyw, 2017-11-29:
would you please post the footprint you loaded in FC? does  it have a value in z rotation angle?

antoniovazquezblanco, 2017-11-29:
No Z rotation. I am starting to use the new https://github.com/KiCad/kicad-packages3D repo with https://github.com/KiCad/kicad-footprints.

The other repos have been declared legacy.

easyw, 2017-11-29:
@terjeio 
would you please have a look at this...
moreover I tried to build a model in FC win64b launching 
`create_model-example.bat`
and I get:

> Exception while processing file: main_generator.py [coercing to Unicode: need string or buffer, NoneType found]

terjeio, 2017-11-29:
I have not made a PR for the updated models yet, the first PR for the generator itself was for the legacy footprints - rotation has obvoiusly changed in the new repository and the script was updated in my latest PR to match that. I'll make a new PR for the updated models, perhaps later today.

I have not updated the example.bat files, sorry for that. A command that works for me is:
```
start "" "C:\Program Files\FreeCAD 0.16\bin\FreeCAD.exe" main_generator.py SW_DIP_x5_W8.61mm_Slide_LowProfile
```
If you still get the error something else is tripping up the code, maybe the UTF-declaration? I am running on Win7 64bit with FreeCad 0.16/CadQuery 1.0.0.

To list the available model names use:
```
start "" "C:\Program Files\FreeCAD 0.16\bin\FreeCAD.exe" main_generator.py list
```
This can be handy to get the names that could be used as basis for (wildcarded) input to the script.

antoniovazquezblanco, 2017-11-29:
Thank you very much to everyone!

easyw, 2017-11-29:
@terjero
'list' is a very handy option...
Please update the bat/sh file, this is giving a simple way to test the generator...
Thx a lot
Maurice

easyw, 2017-11-29:
@terjeio 
still I have an issue launching
`start "" "C:\FreeCAD_016\bin\freecad" main_generator.py SW_DIP_x5_W8.61mm_Slide_LowProfile`

> CadQuery 1.0.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
FC Version 
['0', '16', '6706 (Git)', 'git://github.com/FreeCAD/FreeCAD.git releases/FreeCAD-0-16', '2016/07/02 10:32:16', 'releases/FreeCAD-0-16', 'f86a4e411ff7848dea98d7242f43b7774bee8fa0']FC Version 016
CQ 030 doesn't open example file
Exception while processing file: main_generator.py [coercing to Unicode: need string or buffer, NoneType found]
File format not supported: SW_DIP_x5_W8.61mm_Slide_LowProfile 

I'm on win10 64b and FC0.16-6712 64b

terjeio, 2017-11-29:
@easyw 
I get this when I run the script:
```
CadQuery 1.0.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
FC Version 
['0', '16', '6712 (Git)', 'git://github.com/FreeCAD/FreeCAD.git releases/FreeCAD-0-16', '2017/07/17 17:27:47', 'releases/FreeCAD-0-16', 'da2d364457257a7a8c6fb2137cea12c45becd71a']FC Version 016

Running...
```
When you run the code a exception is triggered for this:
```
try:
    close_CQ_Example(FreeCAD, Gui)
except: # catch *all* exceptions
    print "CQ 030 doesn't open example file"
```
For me it does not. But I have no idea why and if it is relevant, neither do I know why the code is there (in cq_model_generator.py).

Can you try to print something to the console at various places the \_\_init\_\_ method of the ModelGenerator class in cq_model_generator.py?  Does the crash happen there and if so can you narrow it down?
I set up the script environment there and to make it seamless I read from the environment variables KISYS3DMOD and KISYSMOD, perhaps the issue is related to those?

I will update the .bat/.sh files later.

Terje

terjeio, 2017-11-29:
@easyw 
I've found it, the bug is in the \_\_init\_\_ method as I suspected, correct code is:
```python
        self.footprints_dir = os.getenv('KISYSMOD')
        if self.footprints_dir != None and not os.path.isdir(self.footprints_dir):
            self.footprints_dir = None
```
It is due to KISYSMOD not beeing set on your machine. I'll add that to the forthcoming PR.
If you have time can you please verify the fix first?
 
Terje

easyw, 2017-11-29:
Hi Terje,
still not working for me... it is in cq_model_generator.py file?
I don't use the environment variables because not all the users place them outside KiCad (they can be set locally inside KiCad)
P.S. if you can, it is better to use 'is not' instead of '!=' and 'is' instead of '==' for testing an 'if' condition to 'None'

terjeio, 2017-11-30:
@easyw 
I have update the code and made a new PR, hopefully it works now. I need to check out Python idiosyncrasies, there are a few that so far has bitten me.

@antoniovazquezblanco 
3Models should now be ok, a PR was just merged.

antoniovazquezblanco, 2017-11-30:
You guys are awesome! Thanks!

closed 2017-11-30 by antoniovazquezblanco