easyw/kicad-3d-models-in-freecad #129: Geometry Check on BOP

easyw, 2017-12-02:
Hi @Shackmeister and all developers,
I've just updated the **'cq_cad_tools.py'** adding two new functions:
1) checkUnion(doc)
2) checkBOP(shape)

and a source file to run these functions
[checkBOP-example.py](https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/_tools/checkBOP-example.py)
I attached here also a wrong STEP model, not correctly union-ed and with wrong geometries to make some tests.
[resistor-wrong-BOP.zip](https://github.com/easyw/kicad-3d-models-in-freecad/files/1523869/resistor-wrong-BOP.zip)
uploaded also on the repo...

It is suggested to generate/export a STEP model, then close the file generator and import the STEP file;
then run the checkUnion function to see if the STEP file is correcty union-ed in a single part and run the checkBOP function to see if there are errors on the STEP shape.

This is going to be a quite robust check on the generated files... still in same case I found BOP check is not perfect, but it is for sure much better than the previous state.

Shackmeister, 2017-12-02:
this sounds cool :D can you please try and add this in the new Resistor repo? :) would be nice with an example

easyw, 2017-12-02:
@Shackmeister 
now the 'Resistor_SMD' reo has a BOP check
**NB** the BOP check is working **ONLY** with **FC0.17 12703** or later (it will be skipped for previous FC releases)

poeschlr, 2017-12-02:
This means my freecad is too old. (For some reason the last build for fedora was 48 days ago.)
Well i can use my ubuntu vm. Maybe there is a newer version for this operating system.

easyw, 2017-12-02:
@poeschlr 
for BOP check this commit should be enough
https://github.com/FreeCAD/FreeCAD/commit/2a6a58e
but we had some issue with CQ fillet in previous FC0.17 releases...
may be your release could work fine...
you can simple test BOP check with the example files I linked above

easyw, 2017-12-02:
a working models generator with Union and BOP check is ready here:
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/Resistor_SMD

poeschlr, 2017-12-03:
I tested it with your example file and my freecad version indeed discovered that the resistor is invalid.
So i included this test in my updated phoenix contact scripts.