easyw/kicad-3d-models-in-freecad #156: Fix Documentation

esden, 2018-02-25:
I am trying to add new models to the generators. Unfortunately the scripts referred to in the documentation files don't exist. (Like for example `make_gw_export_fc.py`) running the `main_generator.py` scripts that exist inside the deeper directories don't really work, complaining that it can't find the CadQuery module that is definitely installed in my freecad installation. Trying to push this further I ended up with errors complaining that exportVRML module is not present.

I wish to be able to contribute more content to the KiCad 3d models but not being able to figure out how to run the generator scripts makes it very difficult. :(

Maybe there should be a top level make file that generates all the models as every other software project that is expected to run on linux is built? Or at least a toplevel shell script that is tested on the most mundane and wide spread Ubuntu LTS installation with the default FreeCAD V0.15 and CadQuery installed using the addons_installer. 

As a sidenote: Getting that far is already a hugely cumbersome process that requires a multitude of steps to reach, if you want more contributors you will definitely have to simplify this. :(

easyw, 2018-02-25:
please have a look at this reply
https://github.com/easyw/kicad-3d-models-in-freecad/issues/157

FC and CQ are a fast changing environment improving day by day ... it is quite difficult to keep all things in sink, so your contribute would be appreciated

Cheers,
Maurice

easyw, 2018-03-11:
updated requirements

closed 2018-03-11 by easyw