easyw/kicad-3d-models-in-freecad #157: Generators not working on freecad V0.15

esden, 2018-02-25:
According to the README files the scripts should be working with the stable version of freecad that is the default version in Ubuntu aka V0.15.

But cq_cad_tools is checking and trowing an error if the version of freecad is less than V0.17.
https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/_tools/cq_cad_tools.py#L655

You should test the scripts on V0.15, and handle that gracefully unless you want to force everyone who would like to contribute to install developer version of freecad.

poeschlr, 2018-02-25:
freecad 0.15 is hopelessly outdated. For the newer generators you really need 0.17

esden, 2018-02-25:
If that is the case you need to update your documentation and say that potential users/contributors have to install the daily build version of freecad. 

Also v0.15 is totally fine for most of the things, and you have to use something that is commonly in use otherwise you will drive away contributors. I would argue that 0.16 that is currently the "stable" version of freecad is the highest version to use, if you want to attract potential contributors.

easyw, 2018-02-25:
normally FC 0.15 is fine to build the chain, but you canno rely only on FC release version, but you have to match a release of FC with a correct release of CQ...

> I would argue that 0.16 that is currently the "stable" version of freecad is the highest version to use, if you want to attract potential contributors.

You can argue as you want, you cannot contribute at all or/and you can start a new repo with the most advanced README you may produce... it is your time and decision...
But if you want to contribute to this repo you need to change your attitude and may be help in improving what you consider failing your expectations... this is how an open source environment is working... you are not paying for a service here... you are just receiving something very useful for free

Cheers
Maurice

easyw, 2018-03-11:
FC 0.17 is in code freezing and it will be the next stable asap...
then I moved the requirements accordingly

closed 2018-03-11 by easyw