easyw/kicad-3d-models-in-freecad #158: BGA footprint generator failing

esden, 2018-02-25:
I have only tested the BGA generator script for now but it is failing with the "missing CadQuery 0.3.0 or later module". That error is generated despite CadQuery being installed. The actual offending line is `from Gui.Command import *` but it is included in the same try catch block as the CadQuery import command.

This is the offending line: https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/BGA_packages/main_generator.py#L108

Tested with freecad V0.15, V0.16, V0.17 and in all of them this script fails.

easyw, 2018-02-25:
you haven't even cited the CQ release you are using...
The scripts depend on FC and CQ versions... the chain may be or not broken due to recent release of CQ, not tested recently, but anyway IMO, your attitude is not the right one to contribute anyway.

easyw, 2018-03-11:
fixed for FC 0.17 13387 and CQ 1.0.0 [commit](https://github.com/jmwright/cadquery-freecad-module/commit/c23e237bade857ba69a8130d8bdc6d2400e780c6) 
Some of other scripts may also would need the same fix.

closed 2018-03-11 by easyw