easyw/kicad-3d-models-in-freecad #159: [LED_SMD/main_generator.py] No module named exportPartToVRML

M4xTheMan, 2018-03-12:
Hello everybody,

I'm using CadQuery and the script /LED_SMD/main_generator.py for the first time. I'm starting the script by calling `freecad-daily ./Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD/main_generator.py LED_0805`
FreeCAD is starting and I'm getting the following error message `Exception while processing file: ./Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD/main_generator.py [No module named exportPartToVRML]
File format not supported: LED_0805`

Could you please tell me what I'm missing? Thanks a lot!
Max

easyw, 2018-03-12:
Hi @M4xTheMan 
you need to clone the repo to be able to launch the scripts 
Moreover please check to have also CadQuery Module installed for FreeCAD
Maurice

easyw, 2018-03-12:
@Shackmeister 
would you mind to push the SMD LEDs to kicad library?
Thx a lot 😄

M4xTheMan, 2018-03-14:
Thanks for your fast reply. I've downloaded the master branch and extracted it to /home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master - isn't that enough?

I've already ran into the missing CadQuery error and installed it. It's available as a workbench and I tested it with some of the provided examples.

Max

easyw, 2018-03-14:
Hi @M4xTheMan ,
you would need:
1) cadquery WB (switch to CQ WB to check if you have it)
2) python yaml (install it if you don't have it 'sudo pip install pyyaml')
3) cd to the generator folder
`cd /home/max/Downloads/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD/`
4) launch the script through i.e. 
`freecad  /home/max/Downloads/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD/main_generator.py LED_0805_2012Metric`

Shackmeister, 2018-03-14:
finally https://github.com/KiCad/kicad-packages3D/pull/280

M4xTheMan, 2018-03-16:
Thanks again @easyw,

1. I was able to switch to CQ WB
`CadQuery 1.0.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
`
2. I've installed pyyaml
3. done
4. I'm getting following output (+ error) in FreeCAD
```
CadQuery 1.0.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
FC Version 
['0', '17', '13346 (Git)', 'git://github.com/FreeCAD/FreeCAD.git master', '2018/03/01 13:57:00', 'master', '22ee06de8441bfcc9fded2c6de4232d763fb11bc']FC Version 017
../_tools/exportPartToVRML.pyc

Running...
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD/main_generator.pyc
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD
['', '/main_generator.pyc']
LED_SMD
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/
FC Version 
['0', '17', '13346 (Git)', 'git://github.com/FreeCAD/FreeCAD.git master', '2018/03/01 13:57:00', 'master', '22ee06de8441bfcc9fded2c6de4232d763fb11bc']FC Version 017
../_tools/exportPartToVRML.pyc

Running...
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD/main_generator.py
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD
['', '/main_generator.py']
LED_SMD
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/
Exception while processing file: main_generator.py [Fillets requires that edges be selected]
File format not supported: LED_0805_2012Metric 
```

easyw, 2018-03-16:
hi @M4xTheMan 
your error
`Exception while processing file: main_generator.py [Fillets requires that edges be selected]`
reminds me an old release of CQ WB...
could you try to update the CQ WB?
unlikely cq wb doesn't link the release version to commits ... so your CadQuery 1.0.0 could be an old one release...

M4xTheMan, 2018-03-18:
Hey @easyw,
they doesn't link the version release to commits? Great... I've updated CadQuery via
sudo -H pip install cadquery to version 1.1.0.

Unfortunately that didn't make the error message disappear.
```

CadQuery 1.0.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
FC Version 
['0', '17', '13346 (Git)', 'git://github.com/FreeCAD/FreeCAD.git master', '2018/03/01 13:57:00', 'master', '22ee06de8441bfcc9fded2c6de4232d763fb11bc']FC Version 017
../_tools/exportPartToVRML.pyc

Running...
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD/main_generator.pyc
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD
['', '/main_generator.pyc']
LED_SMD
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/
FC Version 
['0', '17', '13346 (Git)', 'git://github.com/FreeCAD/FreeCAD.git master', '2018/03/01 13:57:00', 'master', '22ee06de8441bfcc9fded2c6de4232d763fb11bc']FC Version 017
../_tools/exportPartToVRML.pyc

Running...
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD/main_generator.py
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/LED_SMD
['', '/main_generator.py']
LED_SMD
/home/max/Downloads/FreeCAD/kicad-3d-models-in-freecad-master/cadquery/FCAD_script_generator/
Exception while processing file: main_generator.py [Fillets requires that edges be selected]
File format not supported: LED_0805_2012Metric 
```

easyw, 2018-03-18:
> Great... I've updated CadQuery via
sudo -H pip install cadquery to version 1.1.0.

Sorry I wasn't clear... You need to update the CadQuery FreeCAD WorkBench ...
https://github.com/jmwright/cadquery-freecad-module
You can do it by the Tools Addon manager Menu if you are in FC 0.17

closed 2018-06-12 by easyw