easyw/kicad-3d-models-in-freecad #167: Is there something wrong with the repository?

Misca1234, 2018-05-03:
I am trying to push an update but get an error message

$ git remote -v

origin  https://github.com/Misca1234/kicad-3d-models-in-freecad.git (fetch)
origin  https://github.com/Misca1234/kicad-3d-models-in-freecad.git (push)
upstream        https://github.com/easyw/kicad-3d-models-in-freecad (fetch)
upstream        https://github.com/easyw/kicad-3d-models-in-freecad (push)

$ git push upstream Split_ACDC_DCDC_into_ACDC_and_DCDC_2

remote: Permission to easyw/kicad-3d-models-in-freecad.git denied to Misca1234.
fatal: unable to access 'https://github.com/easyw/kicad-3d-models-in-freecad/': The requested URL returned error: 403

Misca1234, 2018-05-06:
Yes, i know, I have made several pushes before but now it does not work any more

easyw, 2018-05-07:
Hi, may this link could help
https://stackoverflow.com/questions/7438313/pushing-to-git-returning-error-code-403-fatal-http-request-failed

Misca1234, 2018-05-07:
Jee, I found the same link yesterday, I will check up my credentials today to see if it is solved

Misca1234, 2018-05-07:
Do you think there is some clash when i got larger "admin" rights on the kicad repositories ?

Misca1234, 2018-05-07:
i pushed it to my lokal fork instead and was able to push it your

closed 2018-05-07 by Misca1234