easyw/kicad-3d-models-in-freecad #174: Missing first ping marker on SSOP

acolomb, 2018-06-09:
I extended the SSOP generator script to include an SSOP-18 variant, see #173.

Now after proposing it for inclusion in https://github.com/KiCad/kicad-packages3D/pull/320, I noticed that the first pin marker (circle) is missing in the generated FreeCAD model.

I've had a quick look at the source code for ```main_generator.py```, but could not find an obvious reason for it to be missing. There are some commented-out lines, though. Maybe my rather recent FreeCAD version (compared to what some of the READMEs talk about) could play into it?

OS: Ubuntu 18.04 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13522 (Git)
Build type: None
Branch: releases/FreeCAD-0-17
Hash: 3bb5ff4e70c0c526f2d9dd69b1004155b2f527f2
Python version: 2.7.15rc1
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

easyw, 2018-06-10:
@poeschlr 
would you mind to have a look at this issue?
I don't know if this issue may be related to some of latest changes of December 2017
[main_generator.py commits history](https://github.com/easyw/kicad-3d-models-in-freecad/commits/master/cadquery/FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT/main_generator.py
)

Thx

acolomb, 2018-06-11:
Switching to the commented line fixed to generation of the first pin marker. See:

https://github.com/easyw/kicad-3d-models-in-freecad/blob/b689e0b85e66f1328a20b3245b55e0f68e2ea4d7/cadquery/FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT/main_generator.py#L331

For whatever reason that line was commented out? There seems to be LOTS of old testing code cruft in this repo :-)

easyw, 2018-06-11:
> For whatever reason that line was commented out? 

you missed to have a look at the line below 😝 

> There seems to be LOTS of old testing code cruft in this repo :-)

Yes, not really optimized, but useful to generate more than 4K models for KiCAD 😸 
So the issue is still around ...

acolomb, 2018-06-11:
> you missed to have a look at the line below

Which does not produce a first pin marker dot. The old (commented) one does.

easyw, 2018-06-11:
> Which does not produce a first pin marker dot. The old (commented) one does.

then it could be something related to FC version, CQ version or an interaction between them 😃 
One of the nicest part of the mix of CQ and FC is the fast mobility of both main code ....

easyw, 2018-06-11:
@acolomb 
FYI with:
FC OS: Windows 10 - 64-bit Version: 0.17.13510 +12 (Git), OCC version: 7.2.0
and CQ version '1.2.0' I don't have any issue with the pin marker, without the need to change the main-generator code.
the same with FC OS: Windows 10 - 64-bit, Version: 0.18.13920 (Git), OCC version: 7.2.0  and CQ version '1.2.0'
In which OS were you building the model?

acolomb, 2018-06-11:
I'm using Ubuntu 18.04 with the FreeCAD stable PPA. Suppose you mean opencascade? Then it should be 7.2.0. I'm not at the computer right now, can check again tonight.

acolomb, 2018-06-12:
```
ii  freecad                                                          1:0.17.13522-1ppa1~ubuntu18.04.1                       amd64        Extensible Open Source CAx program (beta)
ii  freecad-doc                                                      1:0.17.13522-1ppa1~ubuntu18.04.1                       all          FreeCAD documentation
ii  libocct-data-exchange-7.2:amd64                                  7.2.0-0ppa6~ubuntu18.04.1                              amd64        Open CASCADE Technology module for CAD data format interoperability
ii  libocct-foundation-7.2:amd64                                     7.2.0-0ppa6~ubuntu18.04.1                              amd64        OCCT module underlying all other OCCT classes
ii  libocct-modeling-algorithms-7.2:amd64                            7.2.0-0ppa6~ubuntu18.04.1                              amd64        Open CASCADE Technology geometrical & topological algorithms module
ii  libocct-modeling-data-7.2:amd64                                  7.2.0-0ppa6~ubuntu18.04.1                              amd64        Open CASCADE Technology 2D/3D geometric primitives data structures
ii  libocct-ocaf-7.2:amd64                                           7.2.0-0ppa6~ubuntu18.04.1                              amd64        Open CASCADE Technology application-specific data and functionality
ii  libocct-visualization-7.2:amd64                                  7.2.0-0ppa6~ubuntu18.04.1                              amd64        Open CASCADE Technology graphical data visualization module
```

As stated in my first comment, CadQuery is at 1.2.0. I'm afraid that is completely identical to @easyw's setup, except for the OS.

easyw, 2018-06-12:
@acolomb 
I cannot reproduce your issue with:
OS: Ubuntu 16.04.3 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13522 (Git)
Build type: None
Branch: releases/FreeCAD-0-17
Hash: 3bb5ff4e70c0c526f2d9dd69b1004155b2f527f2
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)
and CadQuery version 1.2.0

acolomb, 2018-06-12:
Seems the only difference is in fact the OS version, see also my initial post in this issue.

How are you calling the scripts? I could not get the `model_filter` (or any other command-line options) to work, so I commented out all except one line in the `all_series` [definition](https://github.com/easyw/kicad-3d-models-in-freecad/blob/b689e0b85e66f1328a20b3245b55e0f68e2ea4d7/cadquery/FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT/main_generator.py#L576).

Then I just executed `freecad main_generator.py` from within the `GW_QFP_SOIC_SSOP_TSSOP_SOT/` directory.

easyw, 2018-06-12:
> Seems the only difference is in fact the OS version, see also my initial post in this issue.

I have no time to test on ubuntu  18.04 ATM
I will do later

> How are you calling the scripts? I could not get the model_filter (or any other command-line options) to work, so I commented out all except one line in the all_series definition.

In the [README.adoc](https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT/README.adoc)
there are some tips...
the use should be like that:
`freecad main_generator.py model_filter=SOIC-8_3.9x4.9mm_Pitch1.27mm`
singolar model

`freecad main_generator.py series=ssop`
all the series

I also noticed that the series works, but not the model_filter as it should
[december 2017 commits](https://github.com/easyw/kicad-3d-models-in-freecad/commits/master/cadquery/FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT/main_generator.py)
@poeschlr would you mind to have a look at?

easyw, 2018-06-18:
@acolomb 
the generator works fine also for single model ... @poeschlr PMd me and here the launching string:
`c:\FreeCAD\bin\freecad main_generator.py  model_filter=SOIC-8_3.9x4.9mm_P1.27mm`
on my Windows main_generator is missing a path, so I'm adding this to the code...
Still I have no way to test your issue on ubuntu 18.04... I'm going to leave it open to see if we can close later on

easyw, 2018-06-18:
@acolomb 
I managed to test it on:
OS: Ubuntu 18.04 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13522 (Git)
Build type: None
Branch: releases/FreeCAD-0-17
Hash: 3bb5ff4e70c0c526f2d9dd69b1004155b2f527f2
Python version: 2.7.15rc1
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

No issue there....
I'm going to close it as resolved.

closed 2018-06-18 by easyw

acolomb, 2018-06-19:
I just tested it again on my machine. Both `model_filter` and the first pin marker gneration (with unmodified script from master) worked.  No idea how it went wrong when first trying, but it definitely was missing the marker.

What I do still get for any command line options supplied to the cadquery script is the following error message from FreeCAD:

```File format not supported: model_filter=SSOP-18_4.4x6.5mm_P0.65mm```

Seems like it tries to parse the option as an additional file name to open.

easyw, 2018-06-19:
> What I do still get for any command line options supplied to the cadquery script is the following error message from FreeCAD:
`File format not supported: model_filter=SSOP-18_4.4x6.5mm_P0.65mm`
Seems like it tries to parse the option as an additional file name to open.

this is a minor warning you are getting by FC after executing the full code, when the arg parser is parsing the last argument...
it does not afflict the execution at all.