easyw/kicad-3d-models-in-freecad #215: Scripts have trouble to handle wide SOIC models

Misca1234, 2018-11-03:
If you add this to the soic parameter file the script will not be able to solve it correclty,
it creates the wrl and step file but when the model is viewed in kicad the top is transparent

    'TSOP-I-32_18.4x8mm_P0.5mm': Params(
        #
        # TSOP I, 32 pins, 18.4x8mm body (https://www.micron.com/~/media/documents/products/technical-note/nor-flash/tn1225_land_pad_design.pdf, http://www.fujitsu.com/downloads/MICRO/fma/pdfmcu/f32pm25.pdf)
        # This model have been auto generated based on the foot print file
        # A number of paramters have been fixed or guessed, such as A2
        # 
        # The foot print that uses this 3D model is TSOP-I-32_18.4x8mm_P0.5mm.kicad_mod
        # 
        the = 9.0,         # body angle in degrees
        tb_s = 0.15,       # top part of body is that much smaller
        c = 0.1,           # pin thickness, body center part height
        R1 = 0.1,          # pin upper corner, inner radius
        R2 = 0.1,          # pin lower corner, inner radius
        S  = 0.1,          # pin top flat part length (excluding corner arc)
#        L = 0.65,         # pin bottom flat part length (including corner arc)
        fp_s = 1,          # True for circular pinmark, False for square pinmark (useful for diodes)
        fp_r = 0.4,        # first pin indicator radius
        fp_d = 0.5,        # first pin indicator distance from edge
        fp_z = 0.05,       # first pin indicator depth
        ef = 0.0,          # fillet of edges  Note: bigger bytes model with fillet
        cc1 = 0.25,        # 0.45 chamfer of the 1st pin corner
        D1 = 8.0,         # body length
        E1 = 18.4,         # body width
        E = 19.9,          # body overall width
        A1 = 0.1,          # body-board separation
        A2 = 1.5,          # body height
        b = 0.2,          # pin width
        e = 0.5,          # pin (center-to-center) distance
        npx = 16,           # number of pins along X axis (width)
        npy = 0,           # number of pins along y axis (length)
        epad = None,       # e Pad
        excluded_pins = None,          # pin excluded
        old_modelName = 'TSOP-I-32_18.4x8mm_P0.5mm',            # modelName
        modelName = 'TSOP-I-32_18.4x8mm_P0.5mm',            # modelName
        rotation = -90,      # rotation if required
        ),

easyw, 2018-11-03:
It may depends on FC, CQ, OCC release version. Would you please resume those? BTW I'm out of the pc now, so I would be able to review it in few days.

easyw, 2018-11-04:
simply change:
`tb_s = 0.15,       # top part of body is that much smaller`
to:
`tb_s = 0.10,       # top part of body is that much smaller`
and the part will be created

Misca1234, 2018-11-05:
Yes, thanks, it works

Complementary push, see
https://github.com/easyw/kicad-3d-models-in-freecad/pull/220

closed 2018-11-05 by Misca1234