easyw/kicad-3d-models-in-freecad #224: DC to DC converter generator no longer works

stambaughw, 2018-11-10:


easyw, 2018-11-10:
@Misca1234
would you please have a look at this issue?
Thx

Misca1234, 2018-11-10:
yes, give me a minute

Misca1234, 2018-11-10:
Hmm, works for me

[image]

CadQuery 1.0.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
FC Version 
['0', '17', '11223 (Git)', 'git://github.com/FreeCAD/FreeCAD.git master', '2017/06/03 11:46:05', 'master', '88bd341ad2da7399719d350bf9f715297d8d9f33']FC Version 017

Running...
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\Converter_DCDC\main_generator.py
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\Converter_DCDC
['', '\\main_generator.py']
Converter_DCDC
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\

Converter_DCDC_XP_POWER_JTExxxxDxx_THT

make_case
make_case_top
make_pins_tht
{(0.8240000009536743, 0.8199999928474426, 0.781000018119812): 'metal grey pins', (0.14800000190734863, 0.14499999582767487, 0.14499999582767487): 'black body'}
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels/Converter_DCDC.3dshapes\Converter_DCDC_XP_POWER_JTExxxxDxx_THT.step exported
License addition successful
######################################################################
[<Part::PartFeature>]
######################################################################
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels/Converter_DCDC.3dshapes\Converter_DCDC_XP_POWER_JTExxxxDxx_THT.wrl written
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels/Converter_DCDC.3dshapes\Converter_DCDC_XP_POWER_JTExxxxDxx_THT.FCStd saved
File format not supported: Converter_DCDC_XP_POWER_JTExxxxDxx_THT

Misca1234, 2018-11-10:
Try

**C:\Program Files\FreeCAD 0.17\bin\FreeCAD.exe' main_generator.py Converter_DCDC_XP_POWER_JTExxxxDxx_THT**

easyw, 2018-11-10:
@Misca1234
Waine is using latest CQ 1.2

Misca1234, 2018-11-10:
I can try to download it but if it is an error in CQ we might have major problems everywhere

easyw, 2018-11-10:
I built recently some models with CQ 1.2 without issues

Misca1234, 2018-11-11:
I installed cadquery 1.2 and it still works for me

........................../FreeCAD_0.17.11223_x64_dev_win/bin/FreeCAD.exe main_generator.py Converter_DCDC_XP_POWER_JTExxxxDxx_THT

[image]

CadQuery 1.0.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
FC Version 
['0', '17', '11223 (Git)', 'git://github.com/FreeCAD/FreeCAD.git master', '2017/06/03 11:46:05', 'master', '88bd341ad2da7399719d350bf9f715297d8d9f33']FC Version 017

Running...
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\Converter_DCDC\main_generator.py
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\Converter_DCDC
['', '\\main_generator.py']
Converter_DCDC
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\

Converter_DCDC_XP_POWER_JTExxxxDxx_THT

make_case
make_case_top
make_pins_tht
{(0.8240000009536743, 0.8199999928474426, 0.781000018119812): 'metal grey pins', (0.14800000190734863, 0.14499999582767487, 0.14499999582767487): 'black body'}
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels/Converter_DCDC.3dshapes\Converter_DCDC_XP_POWER_JTExxxxDxx_THT.step exported
License addition successful
######################################################################
[<Part::PartFeature>]
######################################################################
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels/Converter_DCDC.3dshapes\Converter_DCDC_XP_POWER_JTExxxxDxx_THT.wrl written
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels/Converter_DCDC.3dshapes\Converter_DCDC_XP_POWER_JTExxxxDxx_THT.FCStd saved
File format not supported: Converter_DCDC_XP_POWER_JTExxxxDxx_THT

Misca1234, 2018-11-11:
I have installed the 1.2.0 cadquery into my freecad and it seems to work for me 

[image]

CadQuery 1.2.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
FC Version 
['0', '17', '11223 (Git)', 'git://github.com/FreeCAD/FreeCAD.git master', '2017/06/03 11:46:05', 'master', '88bd341ad2da7399719d350bf9f715297d8d9f33']FC Version 017

Running...
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\Converter_DCDC\main_generator.py
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\Converter_DCDC
['', '\\main_generator.py']
Converter_DCDC
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\

Converter_DCDC_XP_POWER_JTExxxxDxx_THT

make_case
make_case_top
make_pins_tht
{(0.8240000009536743, 0.8199999928474426, 0.781000018119812): 'metal grey pins', (0.14800000190734863, 0.14499999582767487, 0.14499999582767487): 'black body'}
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels/Converter_DCDC.3dshapes\Converter_DCDC_XP_POWER_JTExxxxDxx_THT.step exported
License addition successful
######################################################################
[<Part::PartFeature>]
######################################################################
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels/Converter_DCDC.3dshapes\Converter_DCDC_XP_POWER_JTExxxxDxx_THT.wrl written
C:\Users\stefan\KiCad\NyVersion\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\_3Dmodels/Converter_DCDC.3dshapes\Converter_DCDC_XP_POWER_JTExxxxDxx_THT.FCStd saved
File format not supported: Converter_DCDC_XP_POWER_JTExxxxDxx_THT

stambaughw, 2018-11-11:
I did some more testing and I found that there seems to be an issue on my work laptop with window 10 pro.  On my home system with windows 10 home works fine using the same version of FreeCAD and cadquery.   I will try to figure out why my windows 10 pro system doesn't work when I get a chance.  This issue can be closed.  I will report back if I figure out what went wrong.  Sorry about the noise.

closed 2018-11-11 by easyw

Misca1234, 2018-11-11:
No problem, we are here for you