easyw/kicad-3d-models-in-freecad #296: yaml or not

Misca1234, 2019-06-29:
Hi

What did we say a while ago, was we promoting ".yaml" or "cq_paramter" fileformat format ?

I was looking into "cadquery/FCAD_script_generator/LED_SMD" and it is yaml based regarding push
https://github.com/KiCad/kicad-packages3D/pull/554

easyw, 2019-06-29:
@Misca1234
Till few time ago, yaml was not included on FC Appimage and adding yaml is not easy on Appimages.
Since latest FC 0.19 yaml has been added, so it won't be an issue for next FC releases, but it could for version 0.18 and older.
So yaml could be added for new generators, may be a note could be added to the README of the generator.

Misca1234, 2019-06-29:
Then I wont convert the library and let it be as it is

poeschlr, 2019-06-29:
Yaml will make a lot of sense for things like ic packages ad it would allow data reuse. Would require including the toleranced dimension class from the footprint generator.

easyw, 2019-06-30:
Since yaml is included in FC 0.19 Appimage package, this shouldn't be a stopper anymore.
Then if we want to move to use yaml, because of some enhancement, I think it could be done without any issues. We can just add a remark  of dependencies on the README file.

Misca1234, 2019-06-30:
I will keep the repository as it is 
/kicad-3d-models-in-freecad/cadquery/FCAD_script_generator/LED_SMD

and add 

```
Till few time ago, yaml was not included on FC Appimage and adding yaml is not easy on Appimages.
Since latest FC 0.19 yaml has been added, so it won't be an issue for next FC releases, but it could for version 0.18 and older.

```

To the beginning of the README file

closed 2019-06-30 by Misca1234