easyw/kicad-3d-models-in-freecad #321: Using the scripts with CADquery 1.2.0 and FreeCAD 0.18.3

DanSGiesbrecht, 2019-11-04:
I'm having trouble using the `main_generator.py` script with FreeCAD 0.18.3:

When I run:
```
freecad ./main_generator.py allQFN
```
I get:
```
FreeCAD 0.18.3, Libs: 0.18.3R
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

CadQuery 1.2.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
FC Version 
['0', '18.3', '', 'https://code.launchpad.net/~vcs-imports/freecad/trunk', '2019/07/19 03:17:35']FC Version 018.3
QtCore Version 5.12.2
Python Version 3.7.3 (default, Oct  7 2019, 12:56:13) 
[GCC 8.3.0]
FC Version 
['0', '18.3', '', 'https://code.launchpad.net/~vcs-imports/freecad/trunk', '2019/07/19 03:17:35']FC Version 018.3
QtCore Version 5.12.2
Python Version 3.7.3 (default, Oct  7 2019, 12:56:13) 
[GCC 8.3.0]
Exception while processing file: ./main_generator.py [invalid literal for int() with base 10: '18.3']
```

Is this an issue with CADquery?

easyw, 2019-11-04:
Hi @DanSGiesbrecht 
could you please post your full FC info?
Could you please test with FC 0.18.4?
https://github.com/FreeCAD/FreeCAD/releases/tag/0.18.4

I don't get your FC version

```diff
- '0', '18.3', ''
```

The following are the messages I get in FC 0.18.3 & 0.18.4
<mono>

> import FreeCAD;FreeCAD.Version()
> ['0', '18', '16131 (Git)', 'git://github.com/FreeCAD/FreeCAD.git releases/FreeCAD-0-18', '2019/07/12 21:08:08', 'releases/FreeCAD-0-18', '3129ae4296e40ed20e7b3d460b86e6969acbe1c3']
> 
> import FreeCAD;FreeCAD.Version()
> ['0', '18', '4 (GitTag)', 'git://github.com/FreeCAD/FreeCAD.git releases/FreeCAD-0-18', '2019/10/22 16:53:35', 'releases/FreeCAD-0-18', '980bf9060e28555fecd9e3462f68ca74007b70f8']

</mono>

easyw, 2019-11-05:
@DanSGiesbrecht 
I can replicate the issue on linux... it seems FC 0.18 has messed up the way to format its version string...
I will see if I can workaround it

easyw, 2019-11-05:
@DanSGiesbrecht 
it should be fine now

closed 2019-11-08 by easyw