easyw/kicad-3d-models-in-freecad #322: Using the scripts with CADquery 1.2.0 and FreeCAD 0.18.4

nitrousnrg, 2019-11-11:
Hi Maurice, I'm having troubles getting the script to work in this new computer. I think I was using freecad 17 last time but I couldnt find that version so I'm stuck with fc18.

I'm getting this eError string: "CQ 030 doesn't open example file"

[image]
(sorry, can't figure out the shared clipboard on this VM)

[image]

I have the latest cadquery from the addon manager, a fresh ubuntu install, yaml is installed from pip3.

Not sure what else I can try...

Thanks!

easyw, 2019-11-12:
@nitrousnrg 
could you please specify which model generator are you having issues...
Not all the generators are python3 ready...
Moreover you can copy and paste the error text  message instead of an image

nitrousnrg, 2019-11-12:
Ah, thanks. I tried CP_Radial_THT, LED_SMD and C_Chip_SMD. I'll try to switch the system to python2 if that's the problem.

easyw, 2019-11-12:
The best would be to fix the generator accordingly to the new ones, to be py3 ready 😄 
 https://github.com/easyw/kicad-3d-models-in-freecad/pull/292/files

nitrousnrg, 2019-11-28:
I tried the changes in your link, thanks! Didn't work for me though so I went for a debian VM with pyhon2 freecad package and now its working for me

ugolevant, 2020-03-10:
Hi Maurice,

it sadly seems that the scripts are really not working anymore with latest versions of FreeCAD and CADquery. I've been fighting with that for hours, but almost with no result.

I though managed to bring the Capacitor_THT script to work by commenting the line 114 (# reload(cq_cad_tools)),
but the same workaround doesn't work for CP_Radial_THT script.

I'd very appreciate if you spend some minutes to get it back to work with actual software versions. I think your work is great, your instrument is important and the users should be able to use it for themselves and to contribute.

Please.

God bless.

easyw, 2020-03-11:
@ugolevant 
with https://github.com/easyw/kicad-3d-models-in-freecad/commit/474ab858a077aed577df9fd79ea0047ac9d6773f CP_Radial_THT ias compatible with py3, FC 0.18 & 0.19
For other generator you may apply the same patch and make a PR

evanshultz, 2020-10-17:
I'm getting the same error with FreeCAD 0.18.4 and CQ 1.2.0 and I'm using a script that was recently updated and includes the changes shown above (https://github.com/easyw/kicad-3d-models-in-freecad/pull/292/files and https://github.com/easyw/kicad-3d-models-in-freecad/commit/474ab858a077aed577df9fd79ea0047ac9d6773f) as far as I can see. I didn't find anything else in the issues or PRs which gave me more direction and I'm a newbie to FC.

I'm on Win 10 x64 and trying to run `\kicad-3d-models-in-freecad\cadquery\FCAD_script_generator\Resistor_THT` by calling `
"c:\program files\freecad 0.18\bin\freecad.exe" main_generator.py all`. The same or similar process worked for a number of other scripts, though I was originally using FC 0.17 with all else the same. So what's up with this one?

I'm not sure what the issue is. Sorry for resurrecting this comment but hopefully I'm doing something dumb or this script is broken and it needs to be fixed for everyone and not just me.

Here is the console output:
```
CadQuery 1.2.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
FC Version 
['0', '18', '4 (GitTag)', 'git://github.com/FreeCAD/FreeCAD.git releases/FreeCAD-0-18', '2019/10/22 16:53:35', 'releases/FreeCAD-0-18', '980bf9060e28555fecd9e3462f68ca74007b70f8']FC Version 018
QtCore Version 5.6.2
Python Version 3.6.6 | packaged by conda-forge | (default, Jul 26 2018, 11:48:23) [MSC v.1900 64 bit (AMD64)]

Ex000 Closed
Dock adjusted
CQ 030 doesn't open example fileException while processing file: main_generator.py [No module named 'cq_parameters']
```

chschlue, 2020-10-17:
Note  that `CQ 030 doesn't open example fileException while processing file` is not an error but just a notification that Cadquery 0.3.0 and above doesn't open some example file.
`cq_parameters` not being found isn't related to that.

easyw, 2020-10-17:
@evanshultz 
does it work on FC 0.19?

evanshultz, 2020-10-17:
Thanks for the replies!

I didn't mention that there is no output. The above is all that is shown on the console. No models are created. For me, at least.

`CQ 030 doesn't open example file` comes from line 138  of the script and might not be a big deal. But the exception right after it might be the problem. (Perhaps it would be good to add a newline after each console message?)

I snagged FreeCAD_0.19.22670-Win-Conda_vc14.x-x86_64.7z from https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre and... nope! Same console output and no models generated:
```
07:14:56  CadQuery 1.2.0
CadQuery is a parametric scripting API for creating and traversing CAD models
Author: David Cowden
License: Apache-2.0
Website: https://github.com/dcowden/cadquery
07:14:56  FC Version 
07:14:56  ['0', '19', '22670 (Git)', 'D:\x08ldgit_cachegithub.comFreeCADFreeCAD master', '2020/10/10 09:44:51', 'master', '12155f4aab09047c5697db0c1b3cf93b02edda03']FC Version 019
07:14:56  QtCore Version 5.12.9
07:14:56  Python Version 3.8.6 | packaged by conda-forge | (default, Oct  7 2020, 18:22:52) [MSC v.1916 64 bit (AMD64)]
07:14:56  
Ex000 Closed
07:14:56  Dock adjusted
07:14:56  CQ 030 doesn't open example fileException while processing file: main_generator.py [No module named 'cq_parameters']
```

Could somebody else check if the Resistor_THT script works for them? Maybe it's just me, but I got a bunch of other scripts working just fine so my feeling is that the script is at fault and I'm not familiar with CQ and FC (yet.).

chschlue, 2020-10-17:
Doesn't work for me either (on Linux), but I'm mainly getting Py3 syntax errors.
`importing cq_parameters` works for me though.

Could this be a module search path issue? It looks like the actual file `cq_parameters.py` isn't found after all although it's obviously in your CWD.

evanshultz, 2020-10-19:
Maybe. That seems strange, though.

The THT resistor script was recently updated with the Python 3 stuff I see mentioned above by @Franck78 . So why doesn't it run without Python 3 issues for you? This is not getting any more clear and I was hoping it just ran without any effort.

chschlue, 2020-10-19:
True. My local repo isn't even with upstream although git keeps telling me it is.
Not sure how I screwed it up. Will check later.

Franck78, 2020-10-19:
"c:\program files\freecad 0.18\bin\freecad.exe" main_generator.py all

=> python3  main_generator.py all

but the real one to call is the "create_model.sh" that adds some path yes (I have never opened the .bat version).

Under some linux distro , freecad is 'FreeCAD' and under others 'freecad' meaning you have to fix them for you.

evanshultz, 2020-10-19:
Neither of those works for me (the last one does nothing):
[image]

But even if it did, that's unlike the other generators I used and against the instructions at https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/Resistor_THT, so perhaps both ways should work? Or that doesn't work under Windows?

At any rate, since @Franck78 made the script updates I gotta assume everything works there. I'm still hoping to get confirmation from somebody else if an update-to-date repo runs per the instructions or not. @easyw maybe you have time to check? Or @chschlue if you can get things updated. or anybody else.

Franck78, 2020-10-20:
I have FreeCAD 0.18 only. Forget the 'python main_generator' thing. I was saying wrong.

I think noboby cared for the . bat launcher since a long time.

Not python gourou, but cq_parameter.py being there and valid, it's just search path for python's modules that need adjust.

look the linux laucher: it creates a SCRIPTPATH variable with value = the dir containing cq_parameters.py and probably Freecad use that to extend the 'python path'

```
#!/bin/sh

# Absolute path to this script. /home/user/bin/foo.sh
SCRIPT=$(readlink -f $0)
# Absolute path this script is in. /home/user/bin
SCRIPTPATH=`dirname $SCRIPT`
echo $SCRIPTPATH
cd $SCRIPTPATH
echo Best using FC 0.18
#freecad  main_generator.py  R_Axial_Power_L60.0mm_W14.0mm_P66.04mm
FreeCAD  main_generator.py $1

```

chschlue, 2020-10-20:
I've got the repo issues sorted out and am now getting the same error (`Exception while processing file: main_generator.py [No module named 'cq_parameters']`).

Franck78, 2020-10-20:
try the scripts in WIP directory

It have extra lines not present in the resistor script that modify the script path

```
outdir=os.path.dirname(os.path.realpath(__file__)+"/../_3Dmodels")
scriptdir=os.path.dirname(os.path.realpath(__file__))
sys.path.append(outdir)
sys.path.append(scriptdir)
if FreeCAD.GuiUp:
    from PySide import QtCore, QtGui
```

Franck78, 2020-10-20:
I have added the two lines in the script and I can directly run FreeCad main_generator.py whitout the SCRIPTPATH variable# 

```
maui start
import FreeCAD, Draft, FreeCADGui
import ImportGui
import FreeCADGui as Gui
#from Gui.Command import *

scriptdir=os.path.dirname(os.path.realpath(__file__))
sys.path.append(scriptdir)

if FreeCAD.GuiUp:
    from PySide import QtCore, QtGui

```

evanshultz, 2020-10-20:
@easyw 
It seems the instructions on the Resistor_THT page do not work. Is there some change that needs to be made?

Franck78, 2020-10-20:
scriptdir=os.path.dirname(os.path.realpath(__file__))
sys.path.append(scriptdir)

easyw, 2020-10-21:
> 
> 
> @easyw
> It seems the instructions on the Resistor_THT page do not work. Is there some change that needs to be made?

for me the scripts are running both on Linux and win... Would you mind to post your full FC info? Are you with AppImage?

easyw, 2020-10-21:
@Franck78 
adding to the script the lines you posted
`scriptdir=os.path.dirname(os.path.realpath(__file__))`
`sys.path.append(scriptdir)`
the scripts are running also on FC Conda...
Would you mind to PR the solution?
Thanks
Maurice

Franck78, 2020-10-21:
My version of Python (3.6.10), under linux,  (or is it Freecad 0.18 doing something) does not need the 2 lines.

Patch ready for Resistor_THT and DCDC_Converters which is identical

evanshultz, 2020-10-21:
Confirmed with FC 0.18.4 on Win 10. Adding those two lines to above to main_generator.py works. I put them on lines 69 and 70 bit it shouldn't matter too much. Thanks all!

Franck78, 2020-10-21:
@evanshultz ,
Hello,
if you are working also on the footprints associated with the 3dimages generated here, there is a fix to do in the fp generator

It's about power resistors Vishay CPCX CPCC series
They are similar with two or three footprint with different sizes.
Bug I think, the footpring gen point PDF CPCX but use sizes from CPCC (or vice versa)

So, i added all the two series in the 3d generator but not in the footprint generator.
It's something like 3 missing entries in the equivalent of cq_parameter.

https://www.vishay.com/docs/30116/cpcc.pdf
https://www.vishay.com/docs/30218/cpcx.pdf