easyw/kicad-3d-models-in-freecad #324: Best way to add a new 3D model?

cnieves1, 2019-12-13:
Hi, 
I was trying to use thist tool to generate a 3D model for this PR:
  https://github.com/pointhi/kicad-footprint-generator/pull/452
Which matches the OnSemi Case 506AQ.

I realized that it is already generated with correct exposed pad size if I run "freecad main_generator.py DFN-8-1EP_2x2mm_Pitch0.5mm".

However, Kicad needs a 3D model named as DFN-8-1EP_2x2mm_P0.5mm_EP0.7x1.3mm .
Therefore, what should I do to generate this model (and request a PR afterwards)?

TIA

easyw, 2019-12-13:
Hi @cnieves1 
you would need to modify the following file
https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/QFN_packages/cq_parameters.py#L986
adding the new model parameters.

cnieves1, 2019-12-13:
Hi,
I saw that file, and my first thought was to modify it...

I noticed there are 3D files generated in Housings folder while others are generated in Packages folder.
What is the difference between them?

I think the ones generated in Packages folder follow Kicad naming rules, but I didn't find where they are defined.

Could you tell me more about how these scripts work?

TIA,

cnieves1, 2019-12-13:
In fact, the 3D model I need is the one you point at, bit with a different name... Should I copy all the parameters and change only the name?

easyw, 2019-12-14:
Copying and change what is needed would be the easiest way.

cnieves1, 2019-12-17:
Thanks, created PR:
https://github.com/easyw/kicad-3d-models-in-freecad/pull/325

closed 2019-12-17 by cnieves1