easyw/kicad-3d-models-in-freecad #357: BRepSweep_Translation error with epad line

cnieves1, 2020-04-28:
I was trying to add a 3D model for AVX M620720 antenna, using the following diff: [diff.txt](https://github.com/easyw/kicad-3d-models-in-freecad/files/4544496/diff.txt)
  Datasheet: http://datasheets.avx.com/ethertronics/AVX-E_M620720.pdf

If I uncomment the epad line, I get the following error:
  Exception while processing file: main_generator.py [BRepSweep_Translation::Constructor]
Can anyone help me to know what's wrong with this line? 

Thanks,

easyw, 2020-04-28:
@cnieves1 please refer to the file you are pointing... many folders contain a file with this name

easyw, 2020-05-01:
@cnieves1 
here the correct parameters:
[model-parameters.zip](https://github.com/easyw/kicad-3d-models-in-freecad/files/4565322/model-parameters.zip)

easyw, 2020-05-01:
@cnieves1 
and this is the right model compared to your pdf... particularly the model doesn't have a epad but it has 3 pads.
[model-parameters2.zip](https://github.com/easyw/kicad-3d-models-in-freecad/files/4565676/model-parameters2.zip)

closed 2020-05-02 by easyw

cnieves1, 2020-05-04:
ok, so the problem was A1 can not be zero...
I know the central pad is not an exposed pad, but the problem is that central pin dimensions are different from side pins dimensions, so this was the way I found to generate this model...
I'll do a PR with this.