easyw/kicad-3d-models-in-freecad #373: Exception while processing file: main_generator.py [global name 'math' is not defined]

Qbort, 2020-06-16:
I'm very new to this script so this might be a non-issue and just me being stupid...

I am on Windows 10, FreeCAD version 0.17 and cadquery_module installed from within FreeCAD.
I have opened command prompt and navigated to cadquery\FCAD_script_generator\QFN_packages and run below two commands:
```
taskkill -im freecad.exe /f

start "" "C:\Program Files\FreeCAD 0.17\bin\freecad" main_generator.py QFN-16-1EP_3x3mm_Pitch0.5mm
```

When FreeCAD opens I get below error message in the FreeCAD console:
```
QFN-16-1EP_3x3mm_Pitch0.5mmException while processing file: main_generator.py [global name 'math' is not defined]
File format not supported: QFN-16-1EP_3x3mm_Pitch0.5mm 
```

Is this some bug or is my configuration incorrect?

easyw, 2020-06-16:
@Qbort  I cannot test right now, but would you mind to try with FC0.18.4 or FC0.19 to see if you get the same error?
It should work out of the box.

Qbort, 2020-06-16:
@easyw i just installed FreeCad 0.18.4 (last stable release) and indeed it works perfectly! Well done for all the hard work on this.
I noticed that FreeCad 0.17 uses Python 2, whereas FreeCad 0.18 uses Python 3. 

Issue #315 does mention that Python 2 is now dead, so I my conclusion is that this script doesn't actually need to support early versions.

Can I suggest to add a note on README.MD to say that the script is only supported in FreeCad 0.18 or greater (Python 3 support)? Also there are quite a few other documentation files that might need updating - I can submit a PR if you are okay with it?

Qbort, 2020-06-30:
fixed by pull request #374

closed 2020-06-30 by Qbort