easyw/kicad-3d-models-in-freecad #375: Buzzer_Beeper generator: Black screen in FC 0.18

Qbort, 2020-06-21:
While fixing the Buzzer_Beeper generator in order to work with FC 0.18 (see PR #374), I found that it works in FC 0.17 fine, but when running in FC 0.18, the program freezes (black screen) and nothing else is displayed. 

I have tried to apply typical python3 fixes using 2to3 as well as other manual fixes but none worked (see my last commit [here](https://github.com/easyw/kicad-3d-models-in-freecad/pull/374/commits/12fbaea17d6bc94e08c492c49fc927073fc12b90) with the fixes I tried).

I can't debug this due to FC 0.18 black screening and not showing any exception/error messages - does anyone have any other ideas of what might be causing the behaviour?

It have a feeling that it might not be a python2/3 compatibility issue and instead some sort of issue with FreeCad python API?

Steps to reproduce:

1. cd cadquery/FCAD_script_generator/Buzzer_Beeper
2. start "" "C:\Program Files\FreeCAD 0.18\bin\freecad" main_generator.py MagneticBuzzer_CUI_CST-931RP-A

easyw, 2020-06-21:
@Qbort Have you tried with FC0.19? It could be a driver issue more than a py one.

Qbort, 2020-06-28:
@easyw sorry I didn't get a notification for your comment for some reason. 

I tried FreeCad 0.19 and it crashed immediately upon opening (nothing to do with the script). Afterwards I realised the reason why it crashed was because I use my computer via RDP (I run the computer headless normally and use it from my laptop). FreeCad 0.18 does work via RDP for me, though and most of the other scripts do as well.

TLDR; The above script actually does work in FC 0.18 as well but seems to be incompatible with RDP. I'm closing the issue as I don't think anyone cares about adding support for RDP connections.

closed 2020-06-28 by Qbort