easyw/kicad-3d-models-in-freecad #376: metric fuses

ferdymercury, 2020-07-06:
I was wondering if there is any reason why there are no metric fuses included in the kicad 3d model libraries.

For example, C0402 is generated, but not L0402. Maybe they could be included, giving them a different color? (blue or green)?

ferdymercury, 2020-07-06:
Same with Fuse0402, etc.

easyw, 2020-07-06:
@ferdymercury 
I have no clue on it... probably nobody needed before.
If you just miss something specific, a PR would be welcomed.

ferdymercury, 2020-07-06:
@easyw I just noticed that your script already generates them.

`kicad-3d-models-in-freecad/cadquery/FCAD_script_generator/Fuse $ freecad main_generator_SMD.py all`

Should I post the resulting files as a MR on kicad-3d-models repository, or do you have an automatic synchronization script that pushes any new model into the other repo?

ferdymercury, 2020-07-06:
Ok it seems there is a MR for several months already there: https://github.com/KiCad/kicad-packages3D/pull/672

ferdymercury, 2020-07-06:
Do you have by chance 'merge rights' there?

closed 2020-07-06 by ferdymercury

easyw, 2020-07-08:
@ferdymercury I have no rights there 😺