easyw/kicad-3d-models-in-freecad #389: Problems running the scripts

MountyRox, 2020-09-05:
Hi everyone! Several weeks ago i started contributing to _KiCad/kicad-packeges3D_ and created some 3D files with FreeCAD Part Design. In the discussion of my last PR https://github.com/KiCad/kicad-packages3D-source/pull/293 i understood that a more efficient way is to use the CadQuery scripting. Thereupon i started to get a little bit deeper into this matter and worked through all the examples of CadQuery. But when i tried to understand/run the scipts in this (_easyw/kicad-3d-models-in-freecad_) repository i got a lot of  problems. I could not run any script in FreeCad without getting errors and the same result i get when calling the scripts from the corresponding  dir (e.g. DIP_Parts) 

`...\bin\freecad.exe main_generator.py DIP-10 `

It gives following exception in FreeCAD: 

`Exception while processing file: main_generator.py [Missing parentheses in call to 'print'. Did you mean print("CQ 030 doesn't open example file")? (cq_model_generator.py, line 120)]`

The only success i had with the example file cq-ex2.FCMacro. This file runs without any problems. 

Can anybody give some hints what's going wrong and what i should do to contibute to kicad-3d-models-in-freecad repository sucessfully.

Here my "_About FreeCAD_" output:

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

chschlue, 2020-09-05:
Some of the scripts aren't yet compatible with Python 3.
See https://github.com/easyw/kicad-3d-models-in-freecad/pull/292/files for a typical fix.

MountyRox, 2020-09-05:
@chschlue Do you know if any of the folder is already fixed for Python 3? It would be helpful if i would be able to run at least one script of this repository to get a better understanding of what's going on in the scripts. 
Is it possible to run these scripts on Visual Studio Code to have the ability of debugging?

MountyRox, 2020-09-05:
@chschlue I just cloned a repository from https://github.com/acisternino/kicad-3d-models-in-freecad, branch  _dip-parts-update-to-py3._ Acisternino has updated the scripts in DIP_Parts to Python 3. Now the scripts are running without any error. Thank's for your support.

When starting to write scripts for this repository is it important to be downwards compatible to Python 2?

chschlue, 2020-09-05:
That's for @easyw to decide.

easyw, 2020-09-09:
@MountyRox 
if possible I would like to keep py2 and py3 compatibility... some of recent scripts are already compatible py2&3

MountyRox, 2020-09-10:
@easyw: This raises issues about the procedure to do this. From my limited knowledge i can run Python scripts importing the CadQuery and FreeCAD lib only within FreeCAD and not for example in Visual Studio. And i have no influence which Python version is used by FreeCAD, haven't i?

> Remark:
>  I found a helpful post how to use VS to debug scripts running in FreeCAD:
> (https://forum.freecadweb.org/viewtopic.php?t=28901#:~:text=In%20the%20debug%20menu%20in,you%20would%20normally%20execute%20it.
> But for me this does not solve the above mentioned problem.

easyw, 2020-09-10:
@MountyRox 
1) you can print python error messages in FC Report Panel... this helped me a lot during the debugging without the need of VS ... 
2) the python version is the one bundled with FC (particularly on Win)

MountyRox, 2020-09-10:
@easyw 
Thanks a lot!

closed 2020-09-13 by MountyRox