easyw/kicad-3d-models-in-freecad #406: Resistor_THT script can omit lead bends

evanshultz, 2020-10-22:
I am trying to model a Vishay/Dale CP0003 resistor (https://www.vishay.com/docs/30213/cp.pdf) using the following cq_parameters.py entry:
```
'CP0003': Params(
        body_color_key = 'white body',
        pin_color_key = 'metal grey pins',
        l = 22.22,
        d = 7.94,
        w = 7.94,
        pd = 0.914,
        px = 31.75,
        py = 0.0,
        shape = 'box',
        orient = 'h'
    ),
```

The resulting model has no lead bends nor lead stubs coming out of the body:
[image]

Is there some known and minor fix so that the leads come out properly but the model is still representative of this resistor?

I guess this is a limitation of the script, and different argument values will generate proper leads. If the arguments given are known to produce invalid results, it might be nice to alert the user at a minimum. Preventing model generation would be something else to consider, but certainly alert the user (especially since there isn't a preview for this script like the CP_Radial_THT script) as there's no quick and easy way to see there is a problem during model generation.

In addition, you may notice that the body color isn't very white. I found that other colors work well, and since the color looks more white in PCB design software I suppose this is an issue with FC or a misconfiguration of FC or my PC that is my fault. So I don't *think* there's anything to do about it, but I mention it just in case someone happens to have some ideas and takes the time to share them with me here.

In general, this script doesn't seem to have good error handling or input sanitation. While I was experimenting with various shapes to see what they look like, it wasn't always obvious what parameters were required for each shape (if zero is acceptable or not). It would be helpful to check the arguments based on the shape type and let the user know what's wrong. That all being said, a working generator that does it's thing nicely with a little trial and error is infinitely better than no generator at all!

easyw, 2020-10-24:
@evanshultz 
@Franck78 recently[ updated the script](https://github.com/easyw/kicad-3d-models-in-freecad/pull/402)... may be he can give us some help...

Franck78, 2020-10-24:
About errors on values in scripts: 

bah, for each 3d I made, it was always, try / see result. When there is any error, you immediately get a strange or no 3d image.

The resistor THTscript  is a bit confuse because 'variations' were added, making lots a hard to follow 'if then else'

Start with a resembling model eg

* shape = 'power',
* orient = 'h'
* l=38
* d =w= 9
* pd=1
* px=45
*py=0

You will figure out immediately after that what values to adjust. Promess ;)

Franck78, 2020-10-24:
you will find the color definitions in ../tools/shadersColors.py

Select something else or add a new one.

Franck78, 2020-10-25:
@evanshultz , do you get what you want ?

Franck78, 2020-10-30:
with this:
```
# entries checked against datasheet have a comment with the datasheet link

all_params = {
    'CP003-px45': Params(
        body_color_key = 'white body',
        pin_color_key = 'metal grey pins',
        l = 22.22,
        d = 6.35,
        w = 6.35,
        pd = 0.914,
        px = 45,
        py = 0.0,
        shape = 'power',
        orient = 'h'
    ),
    'CP003-px18': Params(
        body_color_key = 'white body',
        pin_color_key = 'metal grey pins',
        l = 22.22,
        d = 6.35,
        w = 6.35,
        pd = 0.914,
        px = 18,
        py = 0.0,
        shape = 'power',
        orient = 'h'
    ),

}
```
I get that:
[image]

The script is not modifiable for subtle changes like having the power R standing some distance above the board, stand-offs . wire shape specific, offsets specifics for verticals, ....
and the lead are bend because there is no way to install a R without that.

I could do it (more modular and readable) provided you give some feedbacks
@evanshultz 
@easyw 
and  also decide who is the official repo

easyw, 2020-10-30:
@Franck78 green light from me! :smile:

Franck78, 2022-01-08:
@easyw 

Hello Easy,

I did not see a lot of movement on the KiCad 3D models. Barely one added a year ago. 
I'm questioning the usefulness of creating models that are just piling as Issues 'pendind review'.

Anyway I wanted an apparently simple thing to do, put some colors on THT_Resistors. Change colors of some 'faces' to make them 'color coded'.

I did not find anything simple like 

`body = body.union(cq.Workplane("YZ").workplane(offset=offset).circle(params.d/2*0.902).extrude(ring).Color('Red'))
`
and apparently it was a problem in the past : https://github.com/dcowden/cadquery/issues/105

Do you remember anything about that. Is there a way to say "Faces(25).Color = Something" in CQ?

Thank you
Franck